klaasnicolaas/home-assistant-glow #157: Glow calculates too much, what can I do?

Lorthiz, 2022-04-09:
Hello, I am kind of new to this whole HA/ESPHome/monitoring/sensors thing, but I wanted to check how accurate glow is nad from my basic math comparing power on my power counter(the main one mounted by power company) this is what I got:
At midight of 09.04.2022 there was X,
At midnight of 10.04.2022 there was Y,
Difference between X and Y is 10,9kWh, Glow collected data is showing 13,37kWh so that is error of above 20% ;/
Now I am trying to find a cause for that, from what I understand glow is reacting too much and getting recording too many energy. I have photodiode, I set `internal_filter: 100ms`, diode is inside 3dprinted case and I used thin 2sided thin tape to mount it to the power counter - power counter is also in a pretty dark place so there is little chance that glow could fire from sunlight or light bulbs. So I thing that conditions should be correct for it to work properly. One thing that makes me suspicious are duplicated logs in Glow dashboard for total energy. If that is not a problem are there any more settings I can change to make it more accurate? If it fires too often meaby I should increase `internal_filter` even more?

```
00:24:31 | [D] | [sensor:125] | 'House - Power Consumption': Sending state 963.85541 W with 0 decimals of accuracy
-- | -- | -- | --
00:24:31 | [D] | [light:035] | 'Red' Setting:
00:24:31 | [D] | [light:046] | State: ON
00:24:31 | [D] | [sensor:125] | 'House - Daily Energy': Sending state 0.34706 kWh with 3 decimals of accuracy
00:24:31 | [D] | [sensor:125] | 'House - Total Energy': Sending state 36.67900 kWh with 3 decimals of accuracy
00:24:31 | [D] | [sensor:125] | 'House - Total Energy': Sending state 36.67900 kWh with 3 decimals of accuracy
00:24:32 | [D] | [light:035] | 'Red' Setting:
00:24:32 | [D] | [light:046] | State: OFF
00:24:35 | [D] | [sensor:125] | 'House - Power Consumption': Sending state 972.97302 W with 0 decimals of accuracy
00:24:35 | [D] | [light:035] | 'Red' Setting:
00:24:35 | [D] | [light:046] | State: ON
00:24:35 | [D] | [sensor:125] | 'House - Daily Energy': Sending state 0.34806 kWh with 3 decimals of accuracy
00:24:35 | [D] | [sensor:125] | 'House - Daily Energy': Sending state 0.34806 kWh with 3 decimals of accuracy
00:24:35 | [D] | [sensor:125] | 'House - Total Energy': Sending state 36.68000 kWh with 3 decimals of accuracy
00:24:36 | [D] | [light:035] | 'Red' Setting:
00:24:36 | [D] | [light:046] | State: OFF
00:24:39 | [D] | [sensor:125] | 'House - Power Consumption': Sending state 969.30530 W with 0 decimals of accuracy
00:24:39 | [D] | [light:035] | 'Red' Setting:
00:24:39 | [D] | [light:046] | State: ON
00:24:39 | [D] | [sensor:125] | 'House - Daily Energy': Sending state 0.34906 kWh with 3 decimals of accuracy
00:24:39 | [D] | [sensor:125] | 'House - Daily Energy': Sending state 0.34906 kWh with 3 decimals of accuracy
00:24:39 | [D] | [sensor:125] | 'House - Total Energy': Sending state 36.68100 kWh with 3 decimals of accuracy
00:24:39 | [D] | [light:035] | 'Red' Setting:
00:24:39 | [D] | [light:046] | State: OFF
00:24:42 | [D] | [sensor:125] | 'House - Power Consumption': Sending state 971.92224 W with 0 decimals of accuracy
00:24:42 | [D] | [light:035] | 'Red' Setting:
00:24:42 | [D] | [light:046] | State: ON
00:24:42 | [D] | [sensor:125] | 'House - Daily Energy': Sending state 0.35006 kWh with 3 decimals of accuracy
00:24:42 | [D] | [sensor:125] | 'House - Total Energy': Sending state 36.68200 kWh with 3 decimals of accuracy
00:24:43 | [D] | [light:035] | 'Red' Setting:
00:24:43 | [D] | [light:046] | State: OFF
```

Lorthiz, 2022-04-10:
Additional info I did not thought about. I also have solar panels is it possible that energy used from solar panels is triggering a 'blink' while not being calculated by power counter?

banemonster, 2022-05-17:
I have solar panels, my meter shows a solid red light when solar is being exported.
I solved this with the timeout value shown below. It's set to 15 seconds, so the light comes on and remains on, after 15 sec the measurement defaults to 0 and doesn't start displaying usage again until the light starts blinking.

sensor:
  # WiFi signal
  - platform: wifi_signal
    name: "${friendly_name} - WiFi Signal"
    update_interval: 120s

  # Pulse meter
  - platform: pulse_meter
    name: '${friendly_name} - Power Consumption'
    id: sensor_energy_pulse_meter
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    **timeout: 15 sec**
    internal_filter: 200ms
    on_value:
      then:
        - light.turn_on:
            id: led_red
        - delay: 0.5s
        - light.turn_off:
            id: led_red

cliffordwhansen, 2022-06-30:
@Lorthiz I had a similar issue where Glow was reporting my usage as 33KW, after adding the `internal_filter: 200ms` it is now a realistic 1KW

markusg1234, 2022-09-03:
I Have Solar panels also and unfortunately my meter blinks if power is being exported or imported from the grid.

banemonster, 2022-09-03:
> I Have Solar panels also and unfortunately my meter blinks if power is being exported or imported from the grid.

Are you 100% sure it blinks on export?

If I'm exporting it's solid lit, however if I'm exporting for instance 2kw but my house is drawing 2.5kw then the led blinks but only at the rate of 0.5kw import.

markusg1234, 2022-09-04:
> Are you 100% sure it blinks on export?
>
> If I'm exporting it's solid lit, however if I'm exporting for instance 2kw but my house is drawing 2.5kw then the led blinks but only at the rate of 0.5kw [...]

Yeap as my Glow matches the lcd display of current solar output minus what the house is drawing.

I also see the led blinking when the solar is over the house draw or under the house draw.

andrewb2102, 2023-02-11:
> 

Did you find a fix for this?

ahmadyusuf, 2023-04-19:
Try using the left calculation method for the daily power sensor instead of the default, that fixed the issue for me

  # Total day useage
  - platform: total_daily_energy
    name: '${friendly_name} - Daily Energy'
    id: sensor_total_daily_energy
    power_id: sensor_energy_pulse_meter
    unit_of_measurement: 'kWh'
    icon: mdi:circle-slice-3
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    **method: left**
    filters:
      - multiply: 0.001

klaasnicolaas, 2023-04-20:
Is there a somewhat in-depth explanation of what makes the difference in these methods? Because if this makes it easier for people with solar panels to use the Glow, that would be very nice.

ahmadyusuf, 2023-04-20:
@klaasnicolaas here you go:

https://community.home-assistant.io/t/riemann-integral-calculates-wrong-values-with-electrical-devices/328174?fbclid=IwAR3qt3xWZAjT3Ve1-n4zPqyAR7GMxv6Wmx7e6Bh0kgl53BXWSJWV2V3OydU

klaasnicolaas, 2023-04-20:
Maybe not a bad idea to add the `method: left` to the code (but as a comment) and explanation in the FAQ 🤔

sparkyws, 2023-12-07:
GPIO26 fixed all my high readings 

[https://github.com/klaasnicolaas/home-assistant-glow/issues/300](url)

klaasnicolaas, 2024-03-25:
Closing the issue, if your measurements are too high, it is best to first check whether you have properly adjusted the photodiode in terms of sensitivity and, if that does not help, set the `internal_filter` (see also the docs).

closed 2024-03-25 by klaasnicolaas