klaasnicolaas/home-assistant-glow #158: Abnormal value

tiimsvk, 2022-04-10:
Is it possible to limit abnormal values?  Because 10000w at 230v is 43A and that is impossible.
[image]

pacoso81, 2022-04-18:
get reed of the led output. it is using delay function that prevents esp to count pulses.
we need to find a way to implement this using timers for the led to blink.

LarsNorgaard, 2022-04-18:
I think it would help, if we could define how long a pulse usually is.
From what I can see online most meters pulse width is 38ms

So if we could tell esphome that don't count a pulse, if it's shorter than 25ms or longer than 45ms, then I think we wouldn't get those crazy readings.

Source:
https://github.com/Wim3d/Main_power_meter/blob/master/Main_power_meter.ino

klaasnicolaas, 2022-04-18:
I don't think it should be very difficult to filter out those abnormal values, but the problem is that everyone's situation can be slightly different, which makes it difficult to debug things. I've been running this project at home for over 1 year now and I've never had extremely high values.

Maybe using the new option "[internal_filter_mode](https://esphome.io/components/sensor/pulse_meter.html?highlight=internal_filter_mode)" with PULSE can also solve some problems 🤔

pacoso81, 2022-05-19:
i removed LED and switch to pulse mode with 3ms filter. One month without false measurements.

closed 2022-06-28 by github-actions[bot]