klaasnicolaas/home-assistant-glow #160: inflation problem

Jan140cz, 2022-04-10:
INFO Reading configuration /config/esphome/home_assistant_glow.yaml...
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Prague'
INFO Generating C++ source...
INFO Compiling app...
Processing home-assistant-glow (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 3.5.0)

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- 1.2.2
|-- 1.0
|-- 1.0
|-- 1.0
|-- 2.1.0
| |-- 1.2.2
|-- 1.1.0
|-- 1.0
|-- 6.18.5
|-- 1.2.1
Compiling /data/home-assistant-glow/.pioenvs/home-assistant-glow/src/main.cpp.o
Linking /data/home-assistant-glow/.pioenvs/home-assistant-glow/firmware.elf
RAM: [= ] 12.8% (used 42020 bytes from 327680 bytes)
Flash: [===== ] 54.2% (used 994414 bytes from 1835008 bytes)
Building /data/home-assistant-glow/.pioenvs/home-assistant-glow/firmware.bin
esp32_create_combined_bin(["/data/home-assistant-glow/.pioenvs/home-assistant-glow/firmware.bin"], ["/data/home-assistant-glow/.pioenvs/home-assistant-glow/firmware.elf"])
Wrote 0x102ce0 bytes to file /data/home-assistant-glow/.pioenvs/home-assistant-glow/firmware-factory.bin, ready to flash to offset 0x0
======================== [SUCCESS] Took 120.97 seconds ========================
INFO Successfully compiled program.
esptool.py v3.2
Serial port /dev/ttyUSB0
Connecting...........
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 58:bf:25:82:84:ac
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Flash will be erased from 0x00010000 to 0x00102fff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Compressed 994528 bytes to 570070...

A fatal error occurred: Packet content transfer stopped (received 8 bytes)
INFO Upload with baud rate 460800 failed. Trying again with baud rate 115200.
esptool.py v3.2
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 58:bf:25:82:84:ac
Uploading stub...
Running stub...
Stub running...
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Flash will be erased from 0x00010000 to 0x00102fff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Compressed 994528 bytes to 570070...

A fatal error occurred: Packet content transfer stopped (received 8 bytes)

klaasnicolaas, 2022-04-10:
?? Without some additional information we cannot help you further.

Jan140cz, 2022-04-11:
if the student just throws an error at esp

klaasnicolaas, 2022-04-11:
When did it go wrong, what hardware did you have connected?  What have you already tried to solve it yourself?  Again without additional information I cannot help you further.

Jan140cz, 2022-04-11:
broke after inhalation to esp32.  hardware: esp32, 4 - pin photodiode module light brightness detection photoelectric switch intelligent car board
[image]

klaasnicolaas, 2022-04-11:
Have you tried flashing without the photodiode board connected? Flashing may fail as long as it is connected during this process.

Jan140cz, 2022-04-12:
It didn't have to look like everything went well now

Jan140cz, 2022-04-12:
kiz mail so statues you know unavailable. INFO Reading configuration /config/esphome/home_assistant_glow.yaml...
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Prague'
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[image]

Jan140cz, 2022-04-12:
Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[05:03:40][I][logger:214]: Log initialized
[05:03:40][C][status_led:014]: Setting up Status LED...
[05:03:40][C][ota:461]: There have been 5 suspected unsuccessful boot attempts.
[05:03:40][D][esp32.preferences:114]: Saving preferences to flash...
[05:03:40][E][ota:468]: Boot loop detected. Proceeding to safe mode.
[05:03:40][I][app:029]: Running through setup()...
[05:03:40][C][wifi:037]: Setting up WiFi...
[05:03:40][C][wifi:201]: Setting up AP...
[05:03:40][C][wifi:203]:   AP SSID: 'home-assistant-glow'
[05:03:40][C][wifi:204]:   AP Password: ''
[05:03:40][C][wifi:213]:   IP Address: 192.168.4.1
[05:03:40][C][ota:085]: Over-The-Air Updates:
[05:03:40][C][ota:086]:   Address: home-assistant-glow.local:3232
[05:03:40][W][ota:095]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
[05:03:40][I][app:062]: setup() finished successfully!
[05:03:40][I][ota:480]: Waiting for OTA attempt.
[05:03:40][I][app:102]: ESPHome version 2022.3.2 compiled on Apr 12 2022, 04:48:57
[05:03:40][I][app:104]: Project klaasnicolaas.home_assistant_glow version 3.0.0
[05:03:40][C][logger:233]: Logger:
[05:03:40][C][logger:234]:   Level: DEBUG
[05:03:40][C][logger:235]:   Log Baud Rate: 115200
[05:03:40][C][logger:236]:   Hardware UART: UART0
[05:03:40][C][status_led:019]: Status LED:
[05:03:40][C][status_led:020]:   Pin: GPIO5
[05:03:40][C][captive_portal:088]: Captive Portal:
[05:03:40][C][wifi:491]: WiFi:
[05:03:40][C][wifi:353]:   Local MAC: 58:BF:25:82:84:AC
[05:03:40][C][wifi:354]:   SSID: ''[redacted]
[05:03:40][C][wifi:355]:   IP Address: 0.0.0.0
[05:03:40][C][wifi:357]:   BSSID: 00:00:00:00:00:00[redacted]
[05:03:40][C][wifi:358]:   Hostname: 'home-assistant-glow'
[05:03:40][C][wifi:360]:   Signal strength: 0 dB ▂▄▆█
[05:03:40][C][wifi:364]:   Channel: 1
[05:03:40][C][wifi:365]:   Subnet: 0.0.0.0
[05:03:40][C][wifi:366]:   Gateway: 0.0.0.0
[05:03:40][C][wifi:367]:   DNS1: 0.0.0.0
[05:03:40][C][wifi:368]:   DNS2: 0.0.0.0
[05:03:40][C][mdns:084]: mDNS:
[05:03:40][C][mdns:085]:   Hostname: home-assistant-glow
[05:03:40][C][ota:085]: Over-The-Air Updates:
[05:03:40][C][ota:086]:   Address: home-assistant-glow.local:3232
[05:03:40][W][ota:095]: Last Boot was an unhandled reset, will proceed to safe mode in 0 It looks like it's still installed just doesn't connect to Wi-Fi

Jan140cz, 2022-04-12:
thank you for the help everything is fine now

closed 2022-04-12 by klaasnicolaas