klaasnicolaas/home-assistant-glow #164: GPIO12 causes ESP boot issue

clough360, 2022-04-24:
Using GPIO12 for the sensing LED interferes with the boot process as it is a strapping pin. 
The compilation process also warns about this:
[image]

This caused an issue for me where the device would boot only intermittently with GPIO12 connected. When I switched to GPIO13, the issue device booted (and worked) correctly.

I suggest changing the default yaml config file and related documentation to use GPIO13

klaasnicolaas, 2022-04-24:
Seems like a good plan, I have no idea whether GPIO13 is better, but that something needs to be changed, that's one thing for sure.

smarquard, 2022-05-27:
For me GPIO12 caused the OTA update to fail. I switched to GPIO5 which resolved the issue. It would be good to change the default to something else.

anoadragon453, 2022-07-26:
FYI `GPIO15` was also a strapping pin on my board. `GPIO16` was not, however.

PR #228 (2022-11-12): Change default pulse pin
This pull request changes the default pulse pin to a different GPIO, so you no longer get a warning in the ESPHome console.

Fixes #164

_I also want to adjust the GPIO pins for the RGB LED in the future if they give a warning, I'm still looking for feedback/advice which should be the new ones._
closed 2022-11-12 by klaasnicolaas