klaasnicolaas/home-assistant-glow #167: Sensor gone insane

smulle48, 2022-04-27:
Hi

I've setup the HA Glow on my Wemos d1 mini with the standard yams file and it's almost perfect.

I have then used the exact same settings on my NodeMCU esp32-s v1.1 and then the numbers are insane.

The watt is floating around 6000-9000w. the real usage is about 400w.

Is there anything special I have to setup, when using an esp32?

klaasnicolaas, 2022-04-27:
To my knowledge there is nothing special you should set, what if you use a different GPIO pin?

smulle48, 2022-04-27:
> To my knowledge there is nothing special you should set, what if you use a different GPIO pin?

I've tried all possible pins with same result.

I'v just noticed this in the log:

`[22:59:27][D][pulse_meter:061]: No pulse detected for 4294s, assuming 0 pulses/min`

smulle48, 2022-04-28:
Think I found the issue.

Just opposite of what everyone else was saying, my sensor was simply too sensitive. I have never adjusted the trim pot, so I first tried to set it according to what they said. Set it too sensitive and then turn it a bit down.

But in my case (and maybe also other) I found out I had to do the exact opposite. I turned it completely down, and then changed the sensitivity until it just made the sensor register the light. Then I turned it just at bit more sensitive.

With this, all the crazy numbers were gone and it register as it should :-D

closed 2022-06-06 by github-actions[bot]