klaasnicolaas/home-assistant-glow #179: Spikes every 2hrs

DSew20, 2022-07-13:
Hello,

This is a great app and great use of ESPhome.  I have had this running for a few months, and there was always a small periodic spike that I put down to a freezer condenser turning on or something similar.

However, with the recent (very) warm weather, the tape holding the GLOW sensor to the meter flashing LED became unstuck.  So, for a few days the sensor was not picking up any power "flashes".  But, upon inspection I am still seeing the periodic spikes making me wonder if this caused by hardware / firmware / software.

Below is a screenshot.  You can see the readings go from zero for two hours, then a spike, then again, etc.

Any throughs?
  
[image]

DSew20, 2022-07-17:
My latest thinking is could it be the smart electricity meter introducing the spike.  Perhaps when it is sending the electricity usage via WAN to the electricity supplier?  We have a Liberty 100 Smart (rather basic) meter, which uses GPRS to communicate with the supplier.  Could this introduce a spikes as above every 2 hours or so?

IliasEagle, 2022-08-07:
It is noise coming from some device near by. In my case it is the gas boiler.

closed 2022-09-13 by github-actions[bot]