klaasnicolaas/home-assistant-glow #182: magnet?

evlo, 2022-07-23:
How is the 3D printed housing secured to the energy meter? Double sided tape? There is metal (magnetic?) ring on the meter that should be used for these things, right?

slovdahl, 2022-07-28:
I have used double-sided tape between the housing and the meter, and duct tape on it because the double-sided tape was not strong enough to keep it in place 😁 It looks like the picture on https://github.com/klaasnicolaas/home-assistant-glow#home-assistant-glow- is using double-sided tape too.

evlo, 2022-07-30:
Actually i misunderstood this device when I asked the question as it is pulse counter and magnetic mout is used by devices that use optical uart (they usually have both - one ir for data and one visible led for mnpulse)

klaasnicolaas, 2022-07-31:
It is indeed true that it is mounted with double-sided tape and it also depends per meter where exactly you have to connect it. In any case, we do not use the IR interface that has a magnetic ring on it.

closed 2022-08-20 by klaasnicolaas