klaasnicolaas/home-assistant-glow #187: Is there a way to build this on a ESP32-POE?

gcorgnet, 2022-07-28:
Hi,
I am not likely to have an easily accessible power source near my meter (ironic, I know!) and was wondering if PoE would be an option?
(Say, using one of [these ](https://www.aliexpress.com/item/1005003610335294.html?spm=a2g0o.productlist.0.0.b1e847b67tUFvP&algo_pvid=c792e62d-4fdc-41d3-a6e1-05f4d38aade2&algo_exp_id=c792e62d-4fdc-41d3-a6e1-05f4d38aade2-18&pdp_ext_f=%7B%22sku_id%22%3A%2212000026482346796%22%7D&pdp_npi=2%40dis%21NZD%21%2117.24%21%21%213.81%21%21%402101e9d216590342117416227e374b%2112000026482346796%21sea)or similar?)
Thanks

DVehlow, 2022-08-14:
Technically yes you could. But I think the link to the ESP you have shared is not a PoE Esp32. Just one with an ethernet port. 

You could try an ESP32 PoE board like [this](https://www.aliexpress.com/item/1005002110854496.html?spm=a2g0o.productlist.0.0.6b5f47b6lGb0HY&algo_pvid=d7d58fd1-4da0-4f5c-bc51-8f3e64b2c400&algo_exp_id=d7d58fd1-4da0-4f5c-bc51-8f3e64b2c400-14&pdp_ext_f=%7B%22sku_id%22%3A%2212000018772386281%22%7D&pdp_npi=2%40dis%21GBP%2123.27%2118.15%21%21%211.34%21%21%402100bdd716605010621881138eaada%2112000018772386281%21sea&curPageLogUid=fo6cbEjRGKpD). The key to this is to make sure you get a PoE ESP32. Home Assistant Glow should be agnostic to what type of ESP32 you use

klaasnicolaas, 2022-08-17:
Wouldn't know why not, provided you also have the right board and of course you will have to come up with something yourself to mount the ESP on your meter 😉

closed 2022-11-12 by klaasnicolaas