klaasnicolaas/home-assistant-glow #204: Daily Energy resets at 2am

julian100004, 2022-10-04:
The time in the logs is correct, but the daily energy sensor resets every day at 2 a.m. Could someone perhaps help me out here?

klaasnicolaas, 2022-10-04:
Which time platform do you use?

https://esphome.io/components/time.html#home-assistant-time-source

julian100004, 2022-10-05:
Should be homeassistant, since i didn't change it when first flashing the esp.
I tried setting it to sntp, so my file would look like this:

```yaml
substitutions:
  name: home-assistant-glow
packages:
  klaasnicolaas.home-assistant-glow: github://klaasnicolaas/home-assistant-glow/home_assistant_glow.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false

time:
  - platform: sntp
    id: sntp_time

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
````
but whenever i try to flash it wirelessly with the esphome ha integration, i run into errors while it's compiling everything.

klaasnicolaas, 2022-10-21:
Was this already resolved?

julian100004, 2022-10-21:
Sadly not

klaasnicolaas, 2022-10-21:
But are you using the `homeassistant` or the `sntp` time component?

julian100004, 2022-10-24:
I'm trying to figure that out, but i don't know how.
I'm currently trying my way around with esphome, but everything i do just gives me some error.
It probably is still set to homeassistant. So that will probably be the problem. But i have no clue how i can set it to sntp.

klaasnicolaas, 2022-10-24:
You can find that in the yaml file that you used in your own esphome environment for the HA Glow 😉 Maybe you can share what errors you get? Otherwise I can't do much for you to think about how we solve your problem.

> But i have no clue how i can set it to sntp.

Sure? You have already posted the code for that 😉 

```yaml
time:
  - platform: sntp
    id: sntp_time
```

julian100004, 2022-10-24:
Yeah, the problem is, that i set it up using the [github.io](https://klaasnicolaas.github.io/home-assistant-glow/) page. I didn't have to change or set anything up there, because it just worked out of the box like that. 
I then "adopted" it to home assistant via the esphome plugin, but everytime i use the install command, it gives me something like this: [image]

> Sure? You have already posted the code for that 😉

Yes, but i don't know how i can get that specific code onto the esp 😅 and i don't know how i can see what's currently on it.

julian100004, 2022-11-14:
Okay, so i reset the whole thing and started from scratch.

This time i flashed the esp through cmd and the .yaml file, and not the website.
The problem i currently have is, that the sensor blinks, but it doesn't log anything or send anything to has.
[image]

klaasnicolaas, 2022-11-14:
Are you using the correct GPIO pin? I recently changed it to GPIO13

julian100004, 2022-11-14:
My bad, i didn't see that the default pulse_pin got changed.
It works again, now i just have to see if i can fix my problems.

julian100004, 2022-11-14:
> Are you using the correct GPIO pin? I recently changed it to GPIO13

Yes, that was the problem. Thank you

julian100004, 2022-11-15:
So, apart from the time thing, that i think i can fix myself now, i have those huge 3.600.000W spikes :
[image]

I read that some users enable the internal_filter or add some other filter. Could this fix my problem or do you have any idead what i could do?

closed 2022-12-23 by github-actions[bot]