klaasnicolaas/home-assistant-glow #209: Add Bluetooth proxy?

apazosg, 2022-10-14:
Just wondering if it's possible to add the new Bluetooth proxy functionality.
I added:
Bluetooth_proxy:
  Active: true

But it's not getting detected in ha

m-s-e, 2022-10-14:
this is a nice idea. i didn't realise that you could use and ESPhome device (that was doing other things, like the Glow) as a BT proxy.

klaasnicolaas, 2022-10-14:
Sounds like a great addition, I'll give it a try in my own test environment 😄 

If it comes to the point that we add it to the yaml file, then I think it is also useful to make a separate yaml file for it, since for example a wemos / esp8266 does not have bluetooth 🤔

ABredhauer, 2022-10-18:
> Just wondering if it's possible to add the new Bluetooth proxy functionality. I added: Bluetooth_proxy: Active: true
> 
> But it's not getting detected in ha

Based on the code here: https://github.com/esphome/bluetooth-proxies/

I think you also need to add the ble_tracker dependency
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

kirilldem, 2022-10-22:
HA Glow is my first ESPhome device - is it possible to add other sensors on it? can I for example connect a temperature probe to it and report that back to HA? Or scan for my phone's BT signal?

klaasnicolaas, 2022-10-23:
@kirilldem Yes that is all possible

kirilldem, 2022-10-30:
> @kirilldem Yes that is all possible

Thanks! Any pointers where to look or what to search for, googling didn't get me far.

njsaunders, 2022-11-02:
Same here - Just tried to enable Bluetooth proxy but now its ONLY a bluetooth proxy?

Config:

```
---
# Home Assistant Glow
#
# Read your electricity meter by means of the pulse LED on your
# meter, useful if you do not have a serial port (P1).
#
substitutions:
  device_name: home-assistant-glow
  friendly_name: House
  device_description: "Measure your energy consumption with the pulse LED on your smart meter"
  pulse_pin: GPIO12
  status_led: GPIO5
  # imp/kWh rate ⬇ #
  pulse_rate: '1000'

esphome:
  name: '${device_name}'
  comment: '${device_description}'
  project:
    name: "klaasnicolaas.home_assistant_glow"
    version: "3.1.0"
  platform: ESP32
  board: nodemcu-32s

bluetooth_proxy:
  active: true

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: '${device_name}'
    password: !secret fallback_password

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esphome_api_password

ota:
  safe_mode: true
  reboot_timeout: 10min
  num_attempts: 5

web_server:
  port: 80
  auth:
    username: !secret esphome_web_username
    password: !secret esphome_web_password

output:
  # - platform: gpio
  #   pin: GPIO5
  #   id: output_blue
  - platform: gpio
    pin: GPIO2
    id: output_red
  - platform: gpio
    pin: GPIO4
    id: output_green

light:
  - platform: binary
    internal: true
    id: led_red
    name: Red
    output: output_red

# Status LED for connection
status_led:
  pin:
    # Blue LED
    number: ${status_led}

# Sensors for ESP version and WIFI information
text_sensor:
  - platform: version
    hide_timestamp: true
    name: "${friendly_name} - ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} - IP Address"
      icon: mdi:wifi
    ssid:
      name: "${friendly_name} - Connected SSID"
      icon: mdi:wifi-strength-2

sensor:
  - platform: pulse_meter
    name: '${friendly_name} - Power consumption'
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    # internal_filter: 100ms
    on_value:
      then:
        - light.turn_on:
            id: led_red
        - delay: 0.5s
        - light.turn_off:
            id: led_red
    filters:
      # multiply value = (60 / imp value) * 1000
      # - multiply: 60
      - lambda: return x * ((60.0 / ${pulse_rate}) * 1000.0);
    total:
      name: '${friendly_name} - Total energy'
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      state_class: total_increasing
      device_class: energy
      accuracy_decimals: 3
      filters:
        # multiply value = 1 / imp value
        # - multiply: 0.001
        - lambda: return x * (1.0 / ${pulse_rate});
 ```

klaasnicolaas, 2022-11-02:
As far as I know it doesn't mean it's just a bluetooth proxy, it's just an extra option you add to it. So you should also be able to connect other sensors to it. Have decided to not add this as standard to the glow for the time being, this due to the fact that you have to hang the ESP on the laptop to flash if you want to use the bluetooth proxies (can update OTA afterwards).

kirilldem, 2022-11-03:
> As far as I know it doesn't mean it's just a bluetooth proxy, it's just an extra option you add to it. So you should also be able to connect other sensors to it. Have decided to not add this as [...]

Do you mean that you will have to re-flash the device with a cable again? 

Would you mind sharing an optional config of what it would look like, so the adventurous of us can do it :)

ABredhauer, 2022-11-10:
I've succesfully added BT proxy support to my glow by adding this section to the bottom of the YAML. I don't see the HA glow show up as a bluetooth device but looking at some of my ibeacons now show up the glow as the event source so confident its working

```
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

closed 2022-12-18 by github-actions[bot]