klaasnicolaas/home-assistant-glow #213: Power consumption locked at 200kW

forangejuice, 2022-10-19:
I've recently setup my HA glow, all seems to be working, I visually verified that the meter blinks followed by a blink from the sensor so all looking good.
Upon leaving it to sit for a while I noticed the power consumption on the HA dashboard permanently locked at 200,000W, everytime it updates data the number "blinks" slightly but stays solid at 200,000W. I can calculate the actual wattage based on the graph of the total KwH, which comes to around 600W.
I tried turning everything off in my house at the circuit breakers (apart from the router, HA Pi4 and fridge) and let it run for a bit to gather data. It averaged about 300W, I thought this seemed quite high so I ran a kettle which using a Kill-a-watt meter showing its actually using about 3000W but the slope on the KwH graph indicates 5700W usage even after subtracting the "baseline" 300W. 
I'm thinking that the calculations for imp -> KwH and the W calculations are being done wrong and likely linked to each other. I used the exact code from "home_assistant_glow.yaml" as I also have 1000 imp/kWh, only adding things like static IP, secret passwords for wifi, OTA and API keys; Also changing GPIO12 to GPIO16 in substitutions to avoid boot issues.

klaasnicolaas, 2022-10-19:
Looks like the sensor is a bit too sensitive with the led on your meter, have you tried setting the `internal_filter`?

forangejuice, 2022-10-19:
I haven't no, would I be looking to move that value up or down considering the high sensitivity? To be honest I don't quite understand what it does

klaasnicolaas, 2022-10-20:
Only comment out that line and see if that makes a difference, you could also check whether the [photodiode is soldered correctly](https://github.com/klaasnicolaas/home-assistant-glow#wrong-soldered-diode) in terms of polarity. Could you possibly share some of the logs and/or some screenshots? So that we have a better idea of the problem.

forangejuice, 2022-10-20:
Seems to have worked perfectly! I'll do some more tests and try to tune in that filter number but it appears accurate within reason so far. Thanks alot!

closed 2022-10-20 by forangejuice