klaasnicolaas/home-assistant-glow #214: Can´t install with ESPHome

pettib, 2022-10-19:
Tried to do as the documentation show.
I downloaded the yaml file in to ESPHome and edit the wifi settings but always get the error message: Error fetching configuration information.

Any Ide what to look at ?

I have done this before with an older version and that one is still working.

klaasnicolaas, 2022-10-19:
> Error fetching configuration information.

When do you get this message?

klaasnicolaas, 2022-10-20:
You could check if you have set the correct board type, since you probably don't use ESP32S. It could also very well be that this ESP32 is not yet supported.

phantom8, 2022-10-21:
how do i check which board type this esp32 CH9102x has? I'm able to install the generic frimware from ESP web tools and connect to the web interface with no issues.

klaasnicolaas, 2022-10-23:
But then you apparently have an esp8266 and not an esp32? 😅

closed 2022-11-05 by klaasnicolaas