klaasnicolaas/home-assistant-glow #220: No issue, just a huge THANK YOU!!!!

RobertoCravallo, 2022-11-04:
I have been dabbling in HA for a couple of weeks now and am truly impressed! I even bought a powerfox to read my electricity meter. Good hard- and software, really bad data return timing!! So back it went. Stumbled over GLOW!! Hey, I am 66 years young, but I was stumped at first. Never have I tried something like this. Took a while, but I got it up and running. In the how too's, not everything is covered. But I got it up and running!!! Thank you for your work on this and sharing it with us!!!

Regards from Mannheim, Germany

[image]

[image]

klaasnicolaas, 2022-11-04:
Glad to hear you're happy with it.

Could you possibly explain what you think hasn't been covered yet? Because there is always room for improvement. Also funny that you tried the Powerfox before, I wanted to make another integration for that.

./Klaas

RobertoCravallo, 2022-11-04:
> funny that you tried the Powerfox before, I wanted to make another integration for that.
> ./Klaas
Do not bother!! It does work but the hard- and software are very good. powerfox sends data to Berlin and then it is supposed to come back in real time, so you can use it. I'm guessing the powerfox server does not handle all the data to well. I thought I messed up, but it worked early mornings and late at night!!! Once in a while during the day also!!! I have sent it back. Your solution is MUCH better and without depending on anyone!! 

I will write up which problems I had to get it to work a bit later. Have to make lunch for my wife now... ;-)

RobertoCravallo, 2022-11-05:
Problems with realizing GLOW project!

First of all, I am 66 years old and have been around computers for the last 30 years. Who still remembers CLIPPER or DELPHI? I did all right there!!!

So after the data return from powerfox was TOTALLY unreliable, I decided to try my luck with GLOW. I am the type of person, that does not give up, once started!!

Until about three weeks ago, I had never heard of HOME ASSISTANT or ESP32!! ESP32 about a week ago!! I read as much as I could find on the net and ordered the parts from AMAZON. Touch cheaper in China, but I was ready to try this, otherwise I would have kept the powerfox, as it did record the data and only had problems in real time.

Ok so flashed GLOW via your cool WEB based flash and downloaded the yaml.

First problem: Where exactly to put the yaml. After a while, I "discovered" the default yaml. This seems to get overwritten on updating GLOW. So I had to change the 'pulse_rate' back to 10.000 and recompile after the update

klaasnicolaas, 2022-11-05:
> I did not install the extra LED and was very surprised, that the ESP32 board itself makes for a nice light show!!! Not mentioned anywhere. See attached foto, light is shining through the case.

I think this is the power LED of the ESP.

> To flash I had to press one of the buttons on the ESP32

This depends on the type of ESP32 board you use, my experience is that this is not necessary.

> Where exactly to put the yaml?

When you use ESP Webtools it is not necessary to also place the YAML config separately in ESPHome, you can adjust things via the ESPHome UI. I still have to adjust the documentation for that because a lot has changed and there are many different ways to install the firmware on an ESP.

> GPIO12, GPIO4 and GPIO5 have a strange "error" message that I did not understand.

The GPIO pin errors are a known problem, I'm currently testing with other pins and I think one of these days I will change these as default.

Anyway, thanks for your input, I'm going to change some things 😄

RobertoCravallo, 2022-11-05:
> I think this is the power LED of the ESP.
Might be the red one, but it has a second light that blinks intermittently. 

> This depends on the type of ESP32 board you use, my experience is that this is not necessary.
Seems to be common though, as that was how it was portrayed on the site I found the info.

> The GPIO pin errors are a known problem,
GPIO16 seems a good bet, at least in the infos I found.

Gladly I will check / test things, if you need someone to do this!

RobertoCravallo, 2022-11-05:
Could it be, that the powerfox can do this? They offer a model that goes to the front of the meter, the one I had ordered was attached to the top and read the data through the hole in the top.

The one for the front can't work differently than GLOW, or can it?

[image]

klaasnicolaas, 2022-11-05:
I'm not familiar with how powerfox reads this at the top, so can't say how much that is the same or different to how the glow does that.

closed 2022-12-13 by github-actions[bot]