klaasnicolaas/home-assistant-glow #221: sensor.house_power_consumption never negative!!!

RobertoCravallo, 2022-11-05:
After trying to configure my gauges, I got strange results sometimes. 

Now math was never my favorite, but I usually get it to work! I finally noticed, that 'sensor.house_power_consumption' never returns a negative number and it should, as I am pumping a lot of juice into the grid!! Even then I get a positive Nr.!

So my question is this: Should 'sensor.house_power_consumption' not return a negative and not a positive Nr. when juice is being pumped into the grid? Or can't GLOW figure that out?

I need to figure out actual power usage for various states (I can only come up with these three): 'No sun', 'sun > consumption' and 'sun < consumption'.

Is my thinking maybe totally wrong?

klaasnicolaas, 2022-11-05:
That is not how the pulse meter works and differs per type of meter that you have at home.

You have meters that give fewer pulses when you return to the grid or that even the LED lights constantly, other meters have a separate pulse LED for consumption and production (to the grid). So you can't just measure the power flow and show negative values.

./Klaas

RobertoCravallo, 2022-11-05:
So no "fix" for this or is there a different solution?

klaasnicolaas, 2022-11-05:
I don't have a solution for that

RobertoCravallo, 2022-11-06:
To bad, otherwise GLOW is pretty much perfect. 

So maybe you should put a big notice at the top of the GLOW project, that this does not work with installations, that put juice back into the grid. Maybe, I would not have tried it then, but I am still happy that I got it up and running!!! ;-)

klaasnicolaas, 2022-11-06:
That depends very much on your situation, only a meter where both production and consumption of energy happens causes problems.

If you would like to have the value negative, you could also do something with template sensors.

closed 2022-11-12 by klaasnicolaas