klaasnicolaas/home-assistant-glow #225: Dropping connection

ABredhauer, 2022-11-10:
I've had my glow up and running for about a week now and have noticed it will often stop reporting data showing as offline through ESP home and unavailable via Home Assistant. I setup a ping every 30s and can see that even when it is showing as offline it still responds to pings and I've never dropped a packet. 

Any advice on how to help my diagnose this?

klaasnicolaas, 2022-11-10:
Is there a lot of distance between your glow and your router?

Kimble8650, 2022-11-10:
I have the same issue. I've had my glow up and running since January without any bigger hiccups. Now I've noticed that the connection is much more unreliable for maybe a month or so. The only thing that has changed in my setup is the home assistant version.

I have not updated the glow firmware, have not moved it, nor have I moved any wifi access points. Weird.

klaasnicolaas, 2022-11-10:
Strange, I've been running it for years now and never had any problems with the connection. Would therefore not immediately know what this could be, how are the RSSI values?

Kimble8650, 2022-11-10:
Mostly around -67dBm to -76dBm with an occasional outlier. But as said, wifi coverage has never been an issue before. I too am clueless what could be the culprit. I just started investigating this today, so I'm not very far along in the process :D

Kimble8650, 2022-11-14:
Ok, so I have made some progress on my issue
I noticed that my wifi was not as responsive as usual on my other devices, so I started looking at my Unify Controller, and low and behold, suddenly all my APs had been dropped from the controllers sphere of influence.
So I readopted the APs and my wifi experience went up significantly. This did not fix my issue entirely, but it did make a huge difference. See picture below and try to find where I readopted my APs =).

[image]

So for me this has been at least in part, a wifi routing issue. I guess the APs started routing independently on the same SSIDs, which caused a lot of conflicts.

The quest goes on.

closed 2022-12-22 by github-actions[bot]