klaasnicolaas/home-assistant-glow #227: NodeMCUv3 - cannot connect to WiFi - ESPHome is not working

teargas666, 2022-11-12:
Hi, 

I have a problem with using Glow in my Home Asisstant. 
I uploaded it to my ESP8266 NodeMCUv3, but I cannot connect it to WiFi. Below is my Code used in that project. 

`# Home Assistant Glow
#
# Read your electricity meter by means of the pulse LED on your 
# meter, useful if you do not have a serial port (P1).
#
substitutions:
  device_name: ha-glow-uterum
  friendly_name: uterum
  device_description: "Measure your energy consumption with the pulse LED on your smart meter"
  pulse_pin: GPIO12
  status_led: GPIO5
  # imp/kWh rate ⬇ #
  pulse_rate: '1000'
esphome:
  name: '${device_name}'
  comment: '${device_description}'
  project:
    name: "klaasnicolaas.home_assistant_glow"
    version: "2.0.1"
  platform: ESP8266
  board: nodemcuv2
wifi:
  ssid: 'MyWiFiSSID'
  password: 'MyWiFiPassword'
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: 'AP_SSID'
    password: wifi_password
captive_portal:
# Enable logging
logger:
# Enable Home Assistant API
api:
  password: esphome_api_password
ota:
  safe_mode: true
  reboot_timeout: 10min
  num_attempts: 5
web_server:
  port: 80
  auth:
    username: esphome_web_username
    password: esphome_web_password
output:
  # - platform: gpio
  #   pin: GPIO5
  #   id: output_blue
  - platform: gpio
    pin: GPIO2
    id: output_red
  - platform: gpio
    pin: GPIO4
    id: output_green
light:
  - platform: binary
    internal: true
    id: led_red
    name: Red
    output: output_red
# Status LED for connection
status_led:
  pin:
    # Blue LED
    number: ${status_led}
# Sensors for ESP version and WIFI information
text_sensor:
  - platform: version
    hide_timestamp: true
    name: "${friendly_name} - ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} - IP Address"
      icon: mdi:wifi
    ssid:
      name: "${friendly_name} - Connected SSID"
      icon: mdi:wifi-strength-2
sensor:
  - platform: pulse_meter
    name: '${friendly_name} - Power consumption'
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    # internal_filter: 100ms
    on_value:
      then:
        - light.turn_on:
            id: led_red
        - delay: 0.5s
        - light.turn_off:
            id: led_red
    filters:
      # multiply value = (60 / imp value) * 1000
      # - multiply: 60
      - lambda: return x * ((60.0 / ${pulse_rate}) * 1000.0);
    total:
      name: '${friendly_name} - Total energy'
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      state_class: total_increasing
      device_class: energy
      accuracy_decimals: 3
      filters:
        # multiply value = 1 / imp value
        # - multiply: 0.001
        - lambda: return x * (1.0 / ${pulse_rate});`

klaasnicolaas, 2022-11-12:
I don't think it's really useful to throw your wifi credentials on github. See that you are still using an old version of the Home Assistant Glow.

What's in your logs?

teargas666, 2022-11-13:
I've removed it from my post.

I downloaded it from your github

After changing the board it's started to work.

closed 2022-12-21 by github-actions[bot]