klaasnicolaas/home-assistant-glow #231: Working with Solar/PV

ronjtaylor, 2022-11-16:
I have this project working on my Smart Meter but only monitoring LED1.  On my Meter this LED1 pulses when I IMPORT energy from my energy supplier.  LED2 pulses when I export excess energy from my house or solar system.
I read about this project when reading about the Energy Dashboard and falsely made the assumption that it would work for my situation.
My ultimate goal was to use this project to feed into the Energy Dashboard but could not figure out how to do that until I read Discussion #176 and found that I was on the wrong track.
Is it possible to have a second Photodiode on LED2 using the same ESP32 with just different jumpers?
I don't have this level of expertise.

klaasnicolaas, 2022-11-19:
That is certainly possible, with some adjustments you could use 2x the pulse meter component, each for a different pulse LED.

ronjtaylor, 2022-11-19:
I have since had a discussion in HA forum and a kind person explained things this way.

 Consider this scenario...
"Your current LEDs only tell you how much is coming in and how much is going out - to help you better understand assume your solar is generating 1kWh and your house is also using exactly 1kWh and this happens for 6 hours straight - your LEDs will not flash at all during that 6 hour period even though you have generated and consumed 6kWh of Energy."

The terminology used in this project is misleading.  Consumption means just that.  The pulsing LEDS should be correctly titled Import and Export Not HOUSE Consumption.  As my friend explained above in the scenario above.
This project is not suitable for houses with Solar.

klaasnicolaas, 2022-11-19:
I'm curious about this discussion because I don't fully agree with the above scenario. If you have 2 LEDs, they react independently of each other on the import and export. So I'm not quite following that an LED wouldn't flash, then we're talking about a single LED?

About the misleading terms, I can partially agree with that. Home Assistant Glow was originally made for a household without solar panels and purely to measure your house consumption. If you have solar panels, it would only work if you have a separate pulse LED for import/export or if you have a separate kWh meter with a pulse LED. And yes, in the latter situations, the terminology would no longer be correct.

ronjtaylor, 2022-11-19:
Below is a screen dump of a view that I created that I used in
debugging this project.
[image: image.png]

My original intention was to replace the CT Clamp Energyhive  3097 from
EnergyHive (AKA Hildebrand) as this product was phased out of production in
2017.
I read about this project on the blog of the HA Energy Dashboard and got
tunnel vision on what it was for.
I am now going down the Shelly EM path as this measures  consumption with a
CT similar to the Energyhive.

This is what my current Energy Dashboard looks like.  Note how solar and
Consumption are overlapped in Energy usage.
[image: image.png]

closed 2022-12-28 by github-actions[bot]