klaasnicolaas/home-assistant-glow #232: Is Glow redundant? Does it buffer and then submit once WiFi is reestablished?

mortenmoulder, 2022-11-21:
I just installed Glow this evening and it's really precise! So far so good. I tried with ranges of 500 W to 10 kW (plugged in my EV), because I was afraid it would be blinking too fast for it to register. When I got above 10.000 W, it fluxuated between two values (like 10.200 and 10.300), but the meter was showing a different number on the screen, so we'll see in a few days how precise it actually is! I had to add a 100ms filter as well, but that might need to be adjusted.

My question is: What happens if my WiFi turns off? Say I am doing some maintenance on my network. Does it stop reporting all together, or does it collect data and then submit it later? Timestamps don't really matter, as I'm guessing everybody wants average consumption.

Thanks for the great project! 😄

cieska, 2022-11-28:
I would like too that glow has some buffer memory to offload the data in case of home assistant isn't reachable for any reasons. Not sure how that's complicated to program though.

mortenmoulder, 2022-11-28:
@cieska It's really not. If the request you're trying to send fails, save the number to a list. Send the new request, and if it succeeds, try sending the list as well. That's the easy part. The hard part would, probably, be to make something that can calculate the usage based on old values. But then again, shouldn't be too hard.

As I said, I don't care about current consumption. I rather want to know the average consumption 😄 

@klaasnicolaas Thanks for the reply. I didn't see it for some reason. Does that mean that the internal counter on the ESP keeps counting up, and I won't lose any data in HA?

klaasnicolaas, 2022-11-28:
You will get data loss in HA, but the Glow will continue to measure regardless of whether it has lost the WiFi connection.

closed 2023-01-05 by github-actions[bot]