klaasnicolaas/home-assistant-glow #233: sensor.house_daily_energy_cost creating hundreds of thousands of records in MariaDB

stumain, 2022-11-27:
Hi, firstly love this project, was a lot of fun to create this sensor and a great way to get energy consumption when you don't have a smart power meter!  Thank you.

I'm seeing an issue since setting this up.  For some reason my DB started to grow, on investigation I can see most of it is from a single entity_id of sensor.house_daily_energy_cost.  After a recent DB purge and repack it has consumed 20% of the total DB size in just 3 days and the records never seem to get removed without a manual purge or filter.

Am I the only one seeing this large DB growth?  How long is each record supposed to stay in place for?

Not sure what diagnostic info is going to help here, so happy to provide, but looking for where to start...

stumain, 2022-11-28:
Update:  Actually it's all the recording entries for home assistant glow.  Do we need these?  Should I be excluding it from recorder? But wouldn't this break core function of energy usage tracking?  And I would like to see the energy usage over time without needing to move this into influxDB and graphana.

Any help or suggestions appreciated.

[image]

klaasnicolaas, 2022-11-28:
If you would like history of your entities then I would keep it in your database, but have you set up a purge in the recorder integration? Because then it doesn't matter much if a DB becomes large, after x days the oldest data will be deleted again.

It may be that if you exclude the entities that the long-term statistics no longer store the data, so I would be careful with that. If you also want to use influxDB and Grafana, you can do so, because this is separate from the recorder integration and the energy dashboard (in fact, you kind of store the data twice then 😉 ).

stumain, 2022-11-28:
Thanks for the follow up.

Yes, I have a purge of 30 days (ideally I would want 60, but would happily accept 30 for HA sensor history).  However, I've gotten nowhere near that before seeing performance issues because the database has grown to 35 million rows and >12GB in about 12 days. It then gets worse from there because backups take longer (or fail) - in some cases leaving behind large orphaned files further consuming disk space) and history searching slows down across the board when going back to earlier records / ranges.

In fact at one point, the DB was so large I could no longer execute the query in phpMyAdmin to find out what was taking all the space.  The above 279,000 + 225,000 + 224,262 rows shown above is from about 1.5 days of measurements. 

Yes, I figured a recorder exclude of these essentially renders the sensor useless, but I was hoping that perhaps there may have been a direct write of the statistics table instead.  So your saying "exclude the entries from recorder and get no energy dashboard anymore"?

Maybe I was really hoping there was an aggregator task or setting I had missed. e.g. to accept the incoming records initially allowing real time view of power usage as flashes are detected;  but after 1 hour to retain a single row for each second only, after 12 hours to retain a row for each 10 seconds, after 24 hours to retain a row per minute, and so on.  If I understand correctly, this would allow the statistics tables to be populated properly, provide a sufficiently accurate history on power usage of the sensor, but eliminate the millions of rows being stored for every single 1000kwH flash and prevent DB bloat.

Perhaps I'm asking a lot there, but I wanted to check and also be sure the above 800,000+ rows in 36 hours is expected behaviour?

Thanks again!

klaasnicolaas, 2022-11-28:
I only have the purge set to 3 days for the recorder integration, which is more than enough because most people do not look further back and/or the data is no longer relevant after that.

If you really want to build a data history, it is best to throw that into influxDB and selectively with which entities you include (do not use the excluded option).

closed 2023-01-05 by github-actions[bot]

slovdahl, 2023-06-12:
@stumain try with the latest version of Glow and use https://github.com/klaasnicolaas/home-assistant-glow/pull/95, you should at least get less data in HA that way.