klaasnicolaas/home-assistant-glow #240: Issue installing on ESP32

doriangh, 2022-11-30:
Hi!
I am having some trouble installing home-assistant-glow on ESP32.
I have followed all instructions for wiring, I copied the home_assistant_glow.yaml in esphome folder and when I try to install it (either from the ESPHome web dashboard, or directly connected to the raspberry pi), it never finishes the installation, exists with the following error: 

``` *** [/data/home-assistant-glow/.pioenvs/home-assistant-glow/src/main.cpp.o] Error 4 ```

Every time I try installing, I hold the EN button, then hold BOOT button and after that release EN button from the moment I click Install (so before selecting the ESP32 from USB) till when it starts to compile.

I also cannot download the project (as a .bin) to install it on the web interface.
I can configure it using the GitHub page of the repo, but it has an old version of Home-assistant-glow (and I cannot find where to edit the pulse interval)

Here's the log file.
[logs_home_assistant_glow_upload.txt](https://github.com/klaasnicolaas/home-assistant-glow/files/10125264/logs_home_assistant_glow_upload.txt)

jacauc, 2022-11-30:
Issue is likely with your esphome docker instance running out of memory while compiling, not with the ESP32

jacauc, 2022-11-30:
Nice

doriangh, 2022-11-30:
So apparently the PI ran out of memory and either home assistant crashed or the add-on.
But fortunately, there is a flag that can be specified inside the config .yaml of the ESP, under esphome, ``` compile_process_limit: 1 ``` 
 
https://esphome.io/changelog/2022.11.0.html#running-esphome-on-lower-powered-machines

closed 2022-11-30 by doriangh

klaasnicolaas, 2022-11-30:
This information might be useful to add to the FAQ part of the documentation 😉