klaasnicolaas/home-assistant-glow #246: Energy jump at the start of the day since 21/10/22

RATA1, 2022-12-20:
I have had glow running for several months and was working perfectly. Recently I noticed that the energy was incorrect and saw these values in HA;

[image]

Looking back in the history I see that it started to happen overnight on 21/10. Seems I was getting updates of zero regularly but then it stopped and every morning since it jumps from zero to a few hundred Watt hours instantly which then skews the figures.

[image]

Not sure if something in HA broke it or an ESP update.
Anyone else seen this or can shed some light on it?

I have set `internal_filter: 100ms` but this has made no difference.

Thanks

klaasnicolaas, 2022-12-20:
Strange, I've never come across that before

RATA1, 2022-12-20:
Yes, very odd. That is how mine used to look.

Today on looking at the power I get two blips when there was no generation. I use this only for a solar generation meter (as the inverter's energy over-reads) and it was dark at these times.

I have overlaid the inverter's power and it has no spikes but other than that it tracks almost identically.
[image]

closed 2023-03-06 by github-actions[bot]