klaasnicolaas/home-assistant-glow #250: no longer getting any reading after update

hank-droid, 2022-12-31:
Hi,
I started tinkering with this around 1y ago.
I was getting readings in the region of millions, so I parked it until I could print a case, which happened just now.
after powering it with a powerbank, it got connected and I added it in HA via the self discovery (the HA installation is different form the one when i originally installed glow).
I was still getting high readings, so I installed espohome addon to adopt the device in order to add the interval filter.
the glow was adopted, and I have then updated it.

now I no longer get any reading.

any suggestion on what i might have done wrong?

the diode seems to work because the  green led is flashing as picks up light, and i see it flashing differently depending on the sensitivity setting.
the blue led that used to flash on the esp32 does not.

thanks!

```
INFO Reading configuration /config/esphome/home-assistant-glow.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Warsaw'
INFO Generating C++ source...
INFO Compiling app...
Processing home-assistant-glow (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
Library Manager: Installing esphome/Improv @ 1.2.3
INFO Installing esphome/Improv @ 1.2.3
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.3 has been installed!
INFO Improv@1.2.3 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
Compiling /data/home-assistant-glow/.pioenvs/home-assistant-glow/src/esphome/components/api/api_connection.cpp.o
Compiling /data/home-assistant-glow/.pioenvs/home-assistant-glow/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/home-assistant-glow/.pioenvs/home-assistant-glow/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/home-assistant-glow/.pioenvs/home-assistant-glow/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/home-assistant-glow/.pioenvs/home-assistant-glow/src/esphome/components/api/api_server.cpp.o
Compiling /data/home-assistant-glow/.pioenvs/home-assistant-glow/src/esphome/components/api/list_entities.cpp.o
Compiling /data/home-assistant-glow/.pioenvs/home-assistant-glow/src/esphome/components/api/proto.cpp.o
Compiling /data/home-assistant-glow/.pioenvs/home-assistant-glow/src/esphome/components/api/subscribe_state.cpp.o
[... 233 lines cut ...]
[19:05:21][C][mdns:103]: mDNS:
[19:05:21][C][mdns:104]:   Hostname: home-assistant-glow
[19:05:21][C][ota:093]: Over-The-Air Updates:
[19:05:21][C][ota:094]:   Address: home-assistant-glow.local:3232
[19:05:21][C][api:138]: API Server:
[19:05:21][C][api:139]:   Address: home-assistant-glow.local:6053
[19:05:21][C][api:143]:   Using noise encryption: NO
[19:05:21][C][improv_serial:032]: Improv Serial:
[19:05:21][C][wifi_info:009]: WifiInfo IPAddress 'House - IP Address'
[19:05:21][C][wifi_info:009]:   Icon: 'mdi:wifi'
[19:05:21][C][wifi_info:011]: WifiInfo SSID 'House - Connected SSID'
[19:05:21][C][wifi_info:011]:   Icon: 'mdi:wifi-strength-2'
[19:05:22][C][wifi_signal.sensor:009]: WiFi Signal 'House - WiFi Signal'
[19:05:22][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[19:05:22][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[19:05:22][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[19:05:22][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[19:05:24][D][api:102]: Accepted ::FFFF:192.168.1.219
[19:05:25][D][api.connection:918]: Home Assistant 2022.12.8 (::FFFF:192.168.1.219): Connected successfully
[19:05:25][D][time:045]: Synchronized time: 2022-12-31 19:05:25
[19:05:52][D][esp32.preferences:113]: Saving 2 preferences to flash...
[19:05:52][D][esp32.preferences:142]: Saving 2 preferences to flash: 2 cached, 0 written, 0 failed
[19:06:51][D][sensor:127]: 'House - WiFi Signal': Sending state -61.00000 dBm with 0 decimals of accuracy

```

hank-droid, 2022-12-31:
ok I read the changelog and I reverted to the previous gpio pin (12) for sensor and it works. it also seems to have adjusted the rate and seems to be working properly overall!

hank-droid, 2023-01-05:
https://github.com/klaasnicolaas/home-assistant-glow/commit/a09da8f54f993d3239ff113a6e70d7def3b5ebee

[image]

klaasnicolaas, 2023-01-05:
The problem is that there is a difference between the version on the ESPHome webtools site and the yaml file in the repository. The pulse pin change is part of version 4.0.0 and has not yet been "released".

hank-droid, 2023-01-05:
I am not sure how I ended there… just leaving those thoughts here in case they can help someone else troubleshoot.

klaasnicolaas, 2023-01-05:
If you update the Home Assistant Glow in ESPHome, it automatically takes the latest version of the YAML file in this repository, while this is actually more of a development version. It would be better if it looked at the versions associated with a release tag in Github.

This has to be improved, can be improved, but I still have to find out how this can be improved in the field of CI.

Helmasynti, 2023-01-06:
@klaasnicolaas, maybe I don't quite understand what you mean but I downloaded the yaml from here

https://github.com/klaasnicolaas/home-assistant-glow/blob/main/home_assistant_glow.yaml

and the pulse pin change is there. 

It's not a big deal to edit it, but since I am a total novice when it comes to microcontrollers and I built the system more than a year ago, I had forgotten many of the details of how everything was installed etc. It took me an effort to repair the system. Of course, the work was not in vain since now I understand the system better, but it was annoying and a bit scary to realise the system not working after the update. I believe that I am not the only novice around here hence the request to open the issue.

klaasnicolaas, 2023-01-06:
That the yaml file you link to in the main branch is a "development version" and therefore contains (breaking) changes that have not yet been released, or documented in release notes. However the `dashboard_import_url` currently automatically grabs the version from the main branch.

Actually, this is the latest stable version: https://github.com/klaasnicolaas/home-assistant-glow/blob/v3.1.0/home_assistant_glow.yaml

MPJK77, 2023-01-07:
Hi Klaas,

Today I started with the homeassistant glow project, nice work!
The "Get started" link from your github [https://github.com/klaasnicolaas/home-assistant-glow#get-started] brings us to project version "3.1.0" and refers to pulse_pin: GPIO13.

The link you posted in your reaction above yesterday (with v3.1.0 in it) brings us to project version "3.0.0" and refers to pulse_pin: GPIO12.

I am a little bit lost, which one should I use? 

I have installed project version "3.1.0" and connected the photodiode to GPIO13 but I have some strange readings in HA.

In the photo and movie you posted on your github there are four wires going to the photodiode, 
but in the visual schema there are only three wires to the photodiode?
I asume there should be only three wires conform the visual schema: 3v3, gnd, DO, is that wright?

Thank you in advance!

Gr. Maarten

MPJK77, 2023-01-07:
[image]

klaasnicolaas, 2023-01-07:
Understand that it is a bit unclear now and I want to improve that soon. The default branch will be `dev` and `main` a separate branch where I will only merge a stable release into, which I hope will make a better separation and should also be better for those who use updates via the ESPHome web tools.

> The link you posted in your reaction above yesterday (with v3.1.0 in it) brings us to project version "3.0.0" and refers to pulse_pin: GPIO12.

That was a mistake on my part because I had not updated the version in the yaml file itself, in this case the version as tagged with a release is leading. The use of GPIO13 is part of the 4.0.0 release, 3.1.0 still uses GPIO12. If you want to run stable then you should look [here](https://github.com/klaasnicolaas/home-assistant-glow/tree/v3.1.0), if you still want to work with the development version then [here](https://github.com/klaasnicolaas/home-assistant-glow/blob/main/home_assistant_glow.yaml).

> I asume there should be only three wires conform the visual schema: 3v3, gnd, DO, is that wright?

It is true that I have 4 wires going into the photodiode, but A0 is an empty wire that is not used, so in this case also use the schematic shown in the readme.

klaasnicolaas, 2023-01-07:
FYI, I made some changes to the repository that allow for a better separation between what is stable (main branch) and what is development (dev branch). I have also reverted the main branch to how it should be according to the 3.1.0 release and you will therefore continue to use GPIO12 as long as 4.0.0 has not yet been released.

MPJK77, 2023-01-08:
Hi klaas,

Thank you for your work, it is all clear to me now!

klaasnicolaas, 2024-02-12:
Close this issue since v4 is here and the default pulse pin has been changed for ESP32.

closed 2024-02-12 by klaasnicolaas