klaasnicolaas/home-assistant-glow #254: Current Consumption and "Electricity Meter Reading"

Zodiac-69, 2023-01-06:
The formula for Current Consumption - return x * ((60.0 / id(electricity_pulse_rate)) * 1000.0) - did not work for me, it seemed to be out by a factor of 1000.
I used puse rate of 1000/kWh in the global - electricity_pulse_rate
Using this - return x * (3600/60.0) - worked for me

How can i make the "Electricity Meter Reading" count "Down" instead of up.
I want to use it as a Pre-Pay value and subtract from the credit (kWh) i have instead of counting up (Accumulating)

klaasnicolaas, 2023-01-06:
Weird that the formula doesn't work, because whether you do 3600 / 60 or (60 / 1000) * 1000, both come out at 60. Can you share your used yaml code and what your logs measure?

Zodiac-69, 2023-01-06:
I have implemented your code, except for the "GAS" as i only have electricity meter here in South Africa.
The rest is the same using the electricity part only..

# Electricity Pulse meter
  - platform: pulse_meter
    name: 'Current ${friendly_name}'
    id: sensor_electricity_pulse_meter
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin_electricity}
    # internal_filter: 100ms
    filters:
      # multiply value = (60 / imp value) * 1000
      # - multiply: 60
      # "below lambda" - return x * ((60.0 / id(electricity_pulse_rate)) * 1000.0);
      - lambda: |-
          return x * (3600/60.0);

klaasnicolaas, 2023-01-06:
> except for the "GAS"

There is no official support for this at all, unless someone else has modified the code for it 😉 

Do you also have this in your yaml code?
```yaml
number:
  # Select the correct pulse rate for your meter
  - platform: template
    id: select_pulse_rate
    name: 'Puls rate - imp/kWh'
    optimistic: true
    mode: box
    min_value: 100
    max_value: 10000
    step: 100
    restore_value: yes
    initial_value: 1000
```

Zodiac-69, 2023-01-06:
I do, but the code i have looks a bit different that that.
I was using your version 1.2.0

klaasnicolaas, 2023-01-06:
Then that explains why the formula didn't work 😉

Zodiac-69, 2023-01-06:
I shall do a "Clean" setup tomorrow and look at it again.
For now, the Current Consumption is very close to what my inverter's current sensor is indicating.

Zodiac-69, 2023-01-08:
I did a clean setup, and it is working 99% of the time.
Can i set a min / max on the pulse meter watt value?
I am getting very high spikes every now and then - 400,000w spikes
[image]

klaasnicolaas, 2023-01-08:
I think you can set a max, but the best thing is to prevent a peak. Maybe set the photodiode a little less sensitive?

Zodiac-69, 2023-01-08:
I shall check tomorrow morning when i can see better.
I think that the sensitivity might be close to Min. , but i shall confirm.

closed 2023-02-15 by github-actions[bot]