klaasnicolaas/home-assistant-glow #256: Daily Enery Numbers

crcerdas, 2023-01-08:
Guys,

Could you help me if the House Daily Energy numbers are OK?. House power consumption looks good.

[image]

Thnks

klaasnicolaas, 2023-01-08:
The day entity is not correct, it seems way too much compared to your total entity. Could it be that too much was measured in the beginning when adjusting the photodiode?

crcerdas, 2023-01-08:
> The day entity is not correct, it seems way too much compared to your total entity. Could it be that too much was measured in the beginning when adjusting the photodiode?

Probably, i will check tomorrow again.

crcerdas, 2023-01-09:
> The day entity is not correct, it seems way too much compared to your total entity. Could it be that too much was measured in the beginning when adjusting the photodiode?

All seem come back to normal numbers, probably was when adjusting the Photodiode. Thnks for the Work!

closed 2023-01-12 by klaasnicolaas