klaasnicolaas/home-assistant-glow #263: reset of all data

DrJon85, 2023-01-14:
Hi i have recently reinstalled on a new esp32. when setting up initially i have somehow managed to 'use' -117,951.32 kWh.
how do i reset this to 0 please?!

klaasnicolaas, 2023-01-14:
I also asked you for some more information 😉 
Because I don't know exactly what you mean yet, are you talking about one of the entities or a value in the energy dashboard etc.

DrJon85, 2023-01-15:
Hi, yes it is the value in the energy dashboard that has been picked up.
[image]

klaasnicolaas, 2023-01-15:
You can change this in `developer tools` -> `statistics`. On this page, Home Assistant lets you know if there are problems with your statistics and you can correct the value for a specific entity if necessary.

[image]

DrJon85, 2023-01-15:
thankyou, is there a way to delete all data and start fresh?

closed 2023-02-22 by github-actions[bot]