klaasnicolaas/home-assistant-glow #266: Help required to accurately report usage when it drops to Zero

gcorgnet, 2023-01-16:
Hi,

I am a happy user of HA Glow and also have solar panels on my roofs which don't feed into a battery. This means that the power usage which is reported by my meter will sometimes be Zero for periods of time and can very quickly turn to a large ish value (say a cloud passes by)

What happens currently (since there is no LED pulse anymore being shown by the meter) is that the last non Zero value will remain being reported until another pulse comes in, which is misleading.

Is there a way to filter out the sensor so that as time passes without a pulse, the value reported decrease until it reaches 0?

As in if it's reporting 360W of usage, it means my meter is blinking once every 10 secs (assuming a pulse rate of 1000 and that my math is OK). This means that beyond 10s, every seconds that passes without a pulse being detected means that the value being reported should no longer be 360. ie If we reach 20s and still no pulse, we're using 180W (at most), etc...

Could anyone shed any light on the best way to do this?

Thanks

klaasnicolaas, 2023-01-16:
Actually, you can't really use the glow when you have solar panels, the measurements that come out give a distorted picture. But what you might try is the timeout option for the [pulse_meter](https://esphome.io/components/sensor/pulse_meter.html?highlight=pulse_meter) component.

closed 2023-02-24 by github-actions[bot]

slovdahl, 2023-06-12:
> Actually, you can't really use the glow when you have solar panels, the measurements that come out give a distorted picture.

I'm not sure if this entirely accurate. At least my Kamstrup meter seems to only blink for incoming power, not for outgoing. So in my case Glow mostly works fine. But I obviously came here because I have a similar problem with low or zero values taking a long time to show up.

Thanks for the hint about the timeout option, I'll see if it would work out 👍

slovdahl, 2023-06-17:
The timeout helps a bit, but it's not perfect. In an ideal world, the `pulse_meter` sensor could figure out that if more time has passed than the time between the two past pulses, the consumption is going down and it could start outputting lower values every nth second. I started looking into the ESPHome `pulse_meter` code but haven't yet figured out if t's doable.. 😀