klaasnicolaas/home-assistant-glow #270: Daily Energy v. Total Energy

droopycool, 2023-01-25:
Hi, I managed everything working and I think it's great. My only issue is that the "Total Energy" values are not matching the "Daily Energy" ones.

Total energy matches perfectly with the consumption my meter shows, Daily Energy seems to be growing at a much faster rate and not matching or even proportional to the total energy. 

Attached my code (at least the part I think it's relevant) and some captures of the values. Can anybody help me to spot where the issue is?

  # Pulse meter
  - platform: pulse_meter
    name: '${friendly_name} - Power Consumption'
    id: sensor_energy_pulse_meter
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    #internal_filter: 100ms
    on_value:
      then:
        - light.turn_on:
            id: led_red
        - delay: 0.5s
        - light.turn_off:
            id: led_red
    filters:
      # multiply value = (60 / imp value) * 1000
      # - multiply: 60
      - lambda: return x * ((60.0 / id(select_pulse_rate).state) * 1000.0);
    total:
      name: '${friendly_name} - Total Energy'
      id: sensor_total_energy
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-7
      state_class: total_increasing
      device_class: energy
      accuracy_decimals: 3
      filters:
        # multiply value = 1 / imp value
        # - multiply: 0.001
        - lambda: return x * (1.0 / id(select_pulse_rate).state);
  # Total day useage
  - platform: total_daily_energy
    name: '${friendly_name} - Daily Energy'
    id: sensor_total_daily_energy
    power_id: sensor_energy_pulse_meter
    unit_of_measurement: 'kWh'
    icon: mdi:circle-slice-3
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001

# Enable time component to reset energy at midnight
# https://esphome.io/components/time/homeassistant.html
time:
  - platform: homeassistant
    id: homeassistant_time

[image]

[image]

klaasnicolaas, 2023-01-25:
I think it's a distorted view because your total entity already had a value at midnight (what was the value at that time?), then it might be correct.

droopycool, 2023-01-26:
> I think it's a distorted view because your total entity already had a value at midnight (what was the value at that time?), then it might be correct.

To check if we are in the same page: if I take a measurement for total entity at any time of the day minues the value at midnight it should be the same as daily energy, right? It's not working like that for me:

Total Energy @ 00:00: 33.74 KWh
Daily Energy @ 00:00: 0 KWh

Total Energy @ 10:00: 41.06 KWh
Daily Energy @ 10:00: 33.78 KWh

ΔTotal Energy: 7.32 KWh
ΔDaily Energy: 33.78 KWh

[image]

[image]

[image]

klaasnicolaas, 2023-01-31:
I do see what you mean now, has it improved in the meantime?

droopycool, 2023-02-08:
Thanks for the comments. I checked with the logged consumption and the actual meter and I could not detect any power spikes. Also, after it ran several days values started to increase at a similar rate (still not as expected but better).

I reset the ESP32 and started from scratch with new names/entities and have been monitoring for the last 4 days, then both values report the same rate. I suspect that wonky values while setting up for the first time adjusting LED sensitivity could be the culprit here. Either way, thanks for the ideas and support!

closed 2023-02-08 by droopycool