klaasnicolaas/home-assistant-glow #271: Glow sending lots of data

floyd1111, 2023-01-26:
Hi and thanks a lot for this great project! I was able to setup my HA Glow on a ESP32 with this photo diode (https://www.ebay.de/itm/274430239184) and integrated it in to home assistant. I sticked the sensor on my meter (ISKRA MT681) And now it's sending data. A lot of data. :)
On the webserver I can see an output of around 10 values per second for power consumption. The values are quite correct, no strange spikes.
Although I like detailed statistics. I am afraid to kill home assistant with this amount of data. 

Any advice to limit the output to maybe one value every 10 seconds?
Thanks very much for help!

floyd1111, 2023-01-26:
I tried the changes from this pull request, and it seems to be working very well.
https://github.com/klaasnicolaas/home-assistant-glow/pull/269

klaasnicolaas, 2023-01-26:
What pulse rate do you use?

floyd1111, 2023-01-26:
I'm using a rate of 10000.
Right now the pulse meter seems to work very well. The total kwh seems ok too. But the total daily energy sensor stopped working. I added heartbeat: 60s in the filter section of the total daily energy sensor and now it is sending the same value on and on.

floyd1111, 2023-01-27:
Alright, it seems the problem solved itself over night. After the reset of the daily sensor at midnight it's counting correctly. Don't have enough data yet to see how accurate the measurement is compared to the meter, but until now it looks primising. Thanks again for your help and this great little device!

closed 2023-01-31 by klaasnicolaas