klaasnicolaas/home-assistant-glow #273: No ability to update pulse via webserver

eyeisystems, 2023-02-08:
New install of Glow, everything working fine but there's no option to update the pulse value on the webserver? 

[image]

klaasnicolaas, 2023-02-08:
Setting via the UI is part of version 4.0.0 and that is currently still in development. With the current version you are running now, you still have to adjust this in the yaml code.

eyeisystems, 2023-02-09:
Ah no problem, it's just the current readme refers to this so thought i was going mad :)

klaasnicolaas, 2023-02-09:
We have separate branches for what is stable and what is development, I'm still working on streamlining that better, but what you see by default is development.

closed 2023-02-25 by klaasnicolaas