klaasnicolaas/home-assistant-glow #274: Reported Power Usage is higher in the day when I am generating solar but is accurate at night when i am not generating solar

andrewb2102, 2023-02-10:
When I am generating solar power consumption value is reporting the power I am sending back to the grid, not the amount of power I am using. I figured this out by turning on my kettle and the power consumption dropped by 2000w. 

How do I fix this?

Florian-Schmidt, 2023-03-29:
> If you know how much your solar production is, you might be able to make something with a template sensor 🤷‍♂️

Correct, I've got the same situation now and I know the production by the panels but I don't know the consumption of the producted energy. 
So I'm actually testing some template sensors to see if this could be a solution or minimum a workaround.

larseggert, 2023-04-10:
Same issue here - power meter seems to flash the LED also when I am selling power to the grid, which is a bit unexpected. Any ideas for a solution, other than going to something like https://github.com/jomjol/AI-on-the-edge-device?

ahmadyusuf, 2023-04-19:
Try using the left calculation method for the daily power sensor instead of the default, that fixed the issue for me

  # Total day useage
  - platform: total_daily_energy
    name: '${friendly_name} - Daily Energy'
    id: sensor_total_daily_energy
    power_id: sensor_energy_pulse_meter
    unit_of_measurement: 'kWh'
    icon: mdi:circle-slice-3
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    **method: left**
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001

larseggert, 2023-04-19:
Thanks, am trying this now. Where is this documented & what does it to?

Florian-Schmidt, 2023-04-19:
ESP documentation: https://esphome.io/components/sensor/total_daily_energy.html

I try to describe details:
https://community.home-assistant.io/t/esphome-total-daily-energy-sensor-rainfall/339214

ahmadyusuf, 2023-04-19:
It got pointed out to me by someone on the Home Assistant group.

Check out this post
https://community.home-assistant.io/t/riemann-integral-calculates-wrong-values-with-electrical-devices/328174?fbclid=IwAR2w3XyfJh3A3stKCS_wK0cuag5sGcsSZS_xO4t2xZWVfxrVngMc1vB0LX0

The default method fails with systems having solar because the sensor doesn't report anything consumption for sometime and then we it starts reporting, you see a sudden spike in the daily usage because the default calculation method assumes the usage increased gradually from 0 to the current reading over the period of time it was not reporting everything.

Hope it makes sense

larseggert, 2023-04-21:
This doesn't work for me, since the LED on my meter seems to blink both when drawing power rpm the grid and when generating power I to the grid. Anything else I could try? This is my lower meter, FWIW: https://www.echelon.com/assets/bltc1e4347bb1a04c1c/Smart-Meter-MTR-3000-IEC-Poly-Phase-datasheet.pdf

Florian-Schmidt, 2023-04-21:
Changed my ESP configuration yesterday, first check today seems to be working. I’ll monitor it next days.

My alternative before has been a template sensor to calculate the real energy consumption by subtract the estimate self usage with a fixed value. This way wouldn‘t be reliable.

Florian-Schmidt, 2023-05-11:
For me this solutions seems to work. The values are quite near to the reality just a little difference. 

In my case I extended to ESP configuration to have one more sensor with "left" method to be able to compare the behavior and calculate the sold energy. Meanwhile Home Assistant provides many helper function but I didn't find out if there's an easier way to do it.

closed 2023-06-19 by github-actions[bot]