klaasnicolaas/home-assistant-glow #275: Info on reducing glow power usage

nathanhaigh, 2023-02-13:
I run my Glow off a battery pack and would like to see some documentation on how to reduce the power requirements of the Glow. Firstly, it would be good to demonstrate how to disable the various LED indicators to prolong battery life. Also any other thoughts would be great too!

KixMan28, 2023-02-13:
I didn't install any LED to replicate the meter's flashing. Considering that my glow is in the closed electrical box, it has no point to have a blinking LED which is not seen by anyone.

klaasnicolaas, 2023-02-13:
You can omit the status LED 😉

nathanhaigh, 2023-02-16:
Thanks for the hints. I was actually thinking of some explicit instructions for those who might not be as familiar with yaml file changes as yourselves. Making changes to the hardware may not be desirable since the various LED flashes are likely a convenient debug feature that would be nice to enable/disable through the yaml file.

I was also wondering about more advanced ways to save battery drainage e.g. through caching data locally then periodically enabling WiFi, transmitting data before disable WiFi.

klaasnicolaas, 2023-03-26:
> I was also wondering about more advanced ways to save battery drainage e.g. through caching data locally then periodically enabling WiFi, transmitting data before disable WiFi.

I don't think there are many advanced options to deal with that, but then your questions would be more appropriate directly at ESPHome.

closed 2023-04-20 by klaasnicolaas