klaasnicolaas/home-assistant-glow #279: Does v4.0 work?

RamblingGeekUK, 2023-03-15:
I ran version 3.x.x for a while, since launch and since upgrading to version 4, that data coming in is all over the place. Any  suggestions please?

klaasnicolaas, 2023-03-15:
v4 is a development version that is still being worked on.

RamblingGeekUK, 2023-03-15:
Thanks for the reply.  When I tried to go to version 3.x.x.  I just got version 4 again.   What did I miss?  Thanks again.

klaasnicolaas, 2023-03-15:
There have been some adjustments with which version can be found on which branch and the current group of users have a slightly different yaml code, so that updating is based on the default branch in this repository (which is now  the `dev` branch).

RamblingGeekUK, 2023-03-15:
Ah.. I missed that.  I think I understand.  Thank you for your prompt replies and for a great project.

RamblingGeekUK, 2023-03-16:
I tried downloading the yaml from the main branch and the 3.1.1 branch and install and no matter what I do, I end up on version 4.0.0

[image]

klaasnicolaas, 2023-03-16:
Change the package url to: `github://klaasnicolaas/home-assistant-glow/home_assistant_glow.yaml@main`

RamblingGeekUK, 2023-03-16:
oh, doh! Thanks

klaasnicolaas, 2023-03-16:
And there are more things that I actually want to release in the main branch from developtment, but I run into quite a few problems getting the CI/CD properly done with the release drafter and the way updates work in ESPHome. Any help for that is welcome.

RamblingGeekUK, 2023-03-16:
Still no joy. https://www.youtube.com/watch?v=6rp4n62YjxA

RamblingGeekUK, 2023-03-16:
> And there are more things that I actually want to release in the main branch from developtment, but I run into quite a few problems getting the CI/CD properly done with the release drafter and the way updates work in ESPHome. Any help for that is welcome.

I will have a poke around, I have very little knowledge of GitHub Actions and EPSHome but see if I can help.

RamblingGeekUK, 2023-03-16:
Hmm, I guess there is no fix for it then... I will keep on poking, will post here if I find a solution.

JiiTeeAm, 2023-03-24:
I've also installed version 4.0 and can't get i to work. Shows way too big numbers, even when the pulse setting is right. Is there a way to install previous version (manually)? Or some other help is welcome..

klaasnicolaas, 2023-03-25:
You can only downgrade using the yaml file from the main branch

RamblingGeekUK, 2023-03-27:
I have tried this so many times.. never works any more :-( So confused.

klaasnicolaas, 2023-03-27:
Wouldn't really know, would have expected that adding `@main` would ensure that only the main branch is looked at 🤔 Could also just be a bug from ESPHome 🤷🏻‍♂️

RamblingGeekUK, 2023-03-27:
I'm just trying again with a brand new ESP32 device, I have uninstalled EP32Home and reinstalled it.

klaasnicolaas, 2023-03-27:
Will also grab an ESP32 and do some testing

klaasnicolaas, 2023-03-27:
https://youtu.be/6rp4n62YjxA?t=80

Here I chose "Plug in to this computer" and after flashing I am running version 3.1.0

vikohub, 2023-03-30:
> There have been some adjustments with which version can be found on which branch and the current group of users have a slightly different yaml code, so that updating is based on the default branch in this repository (which is now the `dev` branch).

Seems like having dev as the default branch leads to that if one downloads the yaml file manually from github (for initial install), one ends up downloading the 4.0.0 version. Is that on purpose?
I'm not quite sure I understood your comment above, but I'm guessing that has to do with how updating from ESPhome is supposed to work.

klaasnicolaas, 2023-03-30:
It's the friction between how ESPHome handles updates when using `dashboard_import`, setting up a good git flow in this project for stables versions, doing development at the same time and making releases.

At some point, this branch split was chosen because otherwise people would already get breaking changes (such as a default pin switch), while this was something for v4.0 and not for the existing v3.x.

RamblingGeekUK, 2023-03-30:
OK, I'm going to spend some time understanding  more about how ESPHome works with my own test project and see if I can get this figured out... no idea how long it will take

vikohub, 2023-03-30:
Thx @klaasnicolaas .
Just to ensure I got this right and also chose the correct GPIO pin (may help others wondering):

-For those who want to use a stable version now, choose 'main' -> use 3.1.0 -> use pulse_pin: GPIO12

-For those looking to test, choose 'dev' -> use 4.0.0 -> use pulse_pin: GPIO13

Did I get that right?

closed 2023-05-08 by github-actions[bot]