klaasnicolaas/home-assistant-glow #286: "hickup" values on steady light

jzielke84, 2023-04-30:
Hi,

I've installed some solar energy panels which generate more power than I mostly use during normal operation (e.g. injecting the rest into the grid). That causes my Norax 3D mezer to show a steady pulse light (which equals 0 or "no power consumed"). HaGLow also ready this as as 0 but sometimes it causes some "hickups" in the data stream where short spikes are reported out of nowhere. It there any way to compensate this behaviour?

[image]

jzielke84, 2023-04-30:
Pussible dup of https://github.com/klaasnicolaas/home-assistant-glow/issues/284 ?

jzielke84, 2023-04-30:
BTW: I hope I don't have to worry about the exclamation mark on the meter. According to the manual it means "negative results" (e.g. the solar energy being injected).

[image]

klaasnicolaas, 2023-04-30:
Maybe when you try the method option, mentioned [here](https://github.com/klaasnicolaas/home-assistant-glow/issues/274#issuecomment-1514097905)

jzielke84, 2023-04-30:
Thanks @klaasnicolaas, however the graph shown above is showing the actual power consumption, not the total daily energy your comment is referring to.

klaasnicolaas, 2023-05-07:
Actually, the glow is not really suitable for houses with solar panels, the pulse LED therefore behaves in a way that makes measuring difficult.

jzielke84, 2023-06-07:
I’ve switched to the BitShake Smarter Reader so this issue is done.

closed 2023-06-07 by jzielke84