klaasnicolaas/home-assistant-glow #288: Low power mode.

gheydon, 2023-05-12:
Hi,

I am wanting to build this device to monitor my meter. However I do not have access to power at the meter, and can't install a power socket (renting). So I am wanting to run this on a LIPO or a couple of 18650.

So for a start I was thinking of running this with a ESP8266, which should use less power than a ESP32. But what else could I do to keep monitoring running as long as possible.

LankyGit1, 2023-05-24:
Can't you get a long usb cable. I have seen 4m ones on amazon for only £8

Moqume, 2023-06-03:
I've tried the ESP32 on 3x 18650 charged powered by a 4W solar panel, but alas, not enough sun in that location (north facing). Otherwise I can get about 3-5 days out of it without any code change. I've replaced it with  power adapter and extended the cable, which now runs under the door into the cabinet.

You'd want to disable the LED and look at the WiFi power save mode, which on the ESP8266 is set to 'None' by default: https://esphome.io/components/wifi.html#power-save-mode. But keep in mind that it can't really "sleep" much with the monitoring.

closed 2023-07-12 by github-actions[bot]

gheydon, 2023-08-21:
I ended up purchasing an [emerald EMS](https://www.amazon.com.au/Emerald-Home-Electricity-Advisor/dp/B09JFMSJ92/ref=sr_1_1?keywords=emerald+ems&qid=1692586289&sr=8-1) which I can use bluetooth to get the "last mile" (about 2 meters through a wall) and it says the battery will last 7 years.

Also my landlord may freaked out if he saw this attached to the meter is a tidy little unit which he will question but be ok with the emerald EMS.