klaasnicolaas/home-assistant-glow #289: Issues Connecting to ESP

SFreeman87, 2023-05-13:
Loving the idea of this but having a real struggle getting it to work. Been at this on and off for a week now. I've finally managed to get it to register light and send back info via the LOG file (however only when connected via USB using the web.esphome.io browser based installer. Below is a copy of a reset log file from there.

Any help would be greatly appreciated, driving me up the wall running into constant dead ends.. 

`ts Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:259]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:469]: There have been 3 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[D][number:012]: 'Puls rate - imp/kWh': Sending state 1000.000000
[D][number:012]: 'Reset Value - Total Energy kWh': Sending state 0.000000
[C][light:035]: Setting up light 'Red'...
[D][light:035]: 'Red' Setting:
[D][light:040]:   Color mode: 
[D][text_sensor:064]: 'House - ESPHome Version': Sending state '2023.4.4'
[D][sensor:110]: 'House - Daily Energy': Sending state 5.67907 kWh with 3 decimals of accuracy
[C][wifi:038]: Setting up WiFi...
[C][wifi:039]:   Local MAC: 3C:E9:0E:86:2A:BC
[C][wifi:211]: Setting up AP...
[C][wifi:213]:   AP SSID: 'home-assistant-glow'
[C][wifi:214]:   AP Password: ''
[C][wifi:223]:   IP Address: 192.168.4.1
[D][wifi:387]: Starting scan...
[C][web_server:099]: Setting up web server...
[C][ota:093]: Over-The-Air Updates:
[C][ota:094]:   Address: 192.168.1.129:3232
[W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 2 restarts
[C][api:025]: Setting up Home Assistant API server...
[D][wifi:387]: Starting scan...
[I][app:062]: setup() finished successfully!
[D][text_sensor:064]: 'Glow - Installed version': Sending state '4.0.0'
[D][sensor:110]: 'House - WiFi Signal': Sending state 0.00000 dBm with 0 decimals of accuracy
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[D][sensor:110]: 'House - Power Consumption': Sending state 0.00000 W with 0 decimals of accuracy
[D][light:035]: 'Red' Setting:
[D][light:046]:   State: ON
[D][sensor:110]: 'House - Daily Energy': Sending state 5.67907 kWh with 3 decimals of accuracy
[D][sensor:110]: 'House - Total Energy': Sending state 0.00000 kWh with 3 decimals of accuracy
[I][app:102]: ESPHome version 2023.4.4 compiled on May 13 2023, 11:53:55
[I][app:104]: Project klaasnicolaas.home-assistant-glow version 4.0.0
[C][logger:294]: Logger:
[C][logger:295]:   Level: DEBUG
[C][logger:296]:   Log Baud Rate: 115200
[C][logger:297]:   Hardware UART: UART0
[C][status_led:019]: Status LED:
[C][status_led:020]:   Pin: GPIO5
[C][template.number:050]: Template Number 'Puls rate - imp/kWh'
[C][template.number:051]:   Optimistic: YES
[C][template.number:052]:   Update Interval: 60.0s
[C][template.number:050]: Template Number 'Reset Value - Total Energy kWh'
[C][template.number:051]:   Optimistic: YES
[C][template.number:052]:   Update Interval: 60.0s
[C][gpio.output:010]: GPIO Binary Output:
[C][gpio.output:011]:   Pin: GPIO2
[C][gpio.output:010]: GPIO Binary Output:
[C][gpio.output:011]:   Pin: GPIO4
[C][template.text_sensor:021]: Template Sensor 'Glow - Installed version'
[C][template.text_sensor:021]:   Icon: 'mdi:label-outline'
[... 277 lines cut ...]
      device_class: energy
      accuracy_decimals: 3
      filters:
        # multiply value = 1 / imp value
        # - multiply: 0.001
        - lambda: return x * (1.0 / id(select_pulse_rate).state);
  # Total day useage
  - platform: total_daily_energy
    name: '${friendly_name} - Daily Energy'
    id: sensor_total_daily_energy
    power_id: sensor_energy_pulse_meter
    unit_of_measurement: 'kWh'
    icon: mdi:circle-slice-3
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001

# Enable time component to reset energy at midnight
# https://esphome.io/components/time/homeassistant.html
time:
  - platform: homeassistant
    id: homeassistant_time`

SFreeman87, 2023-05-13:
Found a solution.

klaasnicolaas, 2023-05-13:
Still issues?

closed 2023-06-19 by github-actions[bot]