klaasnicolaas/home-assistant-glow #290: esp

cool-irr, 2023-05-17:


klaasnicolaas, 2023-05-17:
I need more information than this 😉

cool-irr, 2023-05-17:
> من به اطلاعات بیشتر از این نیاز دارم😉

What information do you neesd?

klaasnicolaas, 2023-05-17:
Mainly more context, when did you get this, what have you already tried to solve it etc.

Unfortunately, I cannot help you with the above errors alone, because I have no idea where they come from and the relationship to the Home Assistant Glow.

cool-irr, 2023-05-17:
> بیشتر زمینه، چه زمانی این را دریافت کردید، چه چیزی قبلاً برای حل آن تلاش کرده اید و غیره.
> 
> متأسفانه، من نمی‌توانم به تنهایی در مورد خطاهای فوق به شما کمک کنم، زیرا هیچ ایده‌ای ندارم که آنها از کجا می‌آیند و ارتباط آن با Home Assistant Glow چیست.

This program is related to Wi-Fi relay activation.  When I click on upload in Arduino, I get this error

closed 2023-05-17 by klaasnicolaas