klaasnicolaas/home-assistant-glow #291: Glow suddenly stopped working

droopycool, 2023-05-24:
I have been using 2 devices (in 2 different readers) reliably and without issues for a couple of months now. Suddenly on May 17th both of them stopped detecting pulses at the same time. Phisically there has not been any change, I have updated to the latest firmware, they are available in the network, but just not detecting any pulses.

The only event I could find related to this is I think a firmware update was released just in the window when they stopped working. Other than that I have no idea what could have happened. Any ideas?

Below history graph :

[image]

klaasnicolaas, 2023-05-24:
Which version are you running?

droopycool, 2023-05-24:
> Which version are you running?

Firmware: 2023.5.3 (May 23 2023, 19:18:14)
Hardware: 4.0.0

klaasnicolaas, 2023-05-24:
I suspect that during the update you went to the v4.x (development) version and there is a breaking change with a different GPIO pin for the pulse module.

droopycool, 2023-05-24:
> I suspect that during the update you went to the v4.x (development) version and there is a breaking change with a different GPIO pin for the pulse module.

Oh, I totally missed that breaking change... So how would it be best to proceed?

a.- Change the cable going to GPIO12 to GPIO13 physically

or...

b.- Edit the GPIO pin definition from "pulse_pin: GPIO13" to "pulse_pin: GPIO12"?

klaasnicolaas, 2023-05-24:
You couldn't have known that breaking change either, because no v4.x has been released yet, only those who have been running the Home Assistant Glow for a while are missing a piece in their config that should prevent you from accidentally going to the code of the development branch updates.

```yaml
package_import_url: github://klaasnicolaas/home-assistant-glow/home_assistant_glow.yaml@main
```

> So how would it be best to proceed?

My advice would be to use GPIO13 physically.

What may also be different in your config now is the way you set the pulse rate, previously in YAML in development this will now be done in the UI. See also: https://github.com/klaasnicolaas/home-assistant-glow/discussions/222

droopycool, 2023-05-24:
Strange, that line is indeed in my config:

[image]

Anyways, thanks for the support. I will try changing the pin physically later when I get home and hopefully come back with good results (:

droopycool, 2023-05-24:
It turns out it has been plugged in GPIO13 all along...

RamblingGeekUK, 2023-05-25:
Is it working for you now then?   I have been trying to get back to 3.1.0 for a while now, but nothing I do works, I always end up on 4.0.0

droopycool, 2023-05-26:
> 

No, it isn't working yet.

RamblingGeekUK, 2023-05-26:
Could you take a look at #279 which should bring you up to speed? I tried again last night, and still ended up on version 4.0.0, even from my own project fork.

RamblingGeekUK, 2023-06-03:
[image]

RamblingGeekUK, 2023-06-03:
I rebuilt the config manually and flashed, just done this now, currently testing

RamblingGeekUK, 2023-06-04:
Seems to be working again.

[image]

I'm going to do some testing with ESPHome,  I don't fully understand the flow of installing. 

I think the issue with the update maybe something to with the security key.