klaasnicolaas/home-assistant-glow #298: WRONG reading

klaasnicolaas, 2023-07-30:
### Discussed in https://github.com/klaasnicolaas/home-assistant-glow/discussions/297

<div type='discussions-op-text'>

<sup>Originally posted by **stackner** July 30, 2023</sup>
have been using this for a couple of months now and have to set to 1000 Puls rate - imp/kWh as is on my meter.
however i seen to sometimes read the correct energy "i think" but sometimes read totally wrong like say 5kwh used in one hour at 1am when nothing is being used in the house. 
This is in the home assistant energy dashboard. i have the sensor for that set as the Total Energy sensor.
</div>

klaasnicolaas, 2023-07-30:
Could it be that you have adjusted the sensor a bit too sensitive?  Have you compared the number of kWh with what your meter indicates and the glow after 1 day?

stackner, 2023-08-02:
how do you adjust the sensitivity? i see no option in your yaml?

klaasnicolaas, 2023-08-02:
You can change it on the PCB board of the photodiode, you will find a blue square and with a screwdriver you can adjust the sensitivity

stackner, 2023-08-07:
my total DAILY usage seems correct it's just the overall TOTAL is too much.
Am i missing something in settings? for instance my daily might show 6kwh but in the TOTAL it might show 4kw for one hour and 12kw for another and 9kw for another and end up being like 40kwh for a day when it only shows 6kwh in the daily meter?

stackner, 2023-08-07:
due to this i don't think it is the sensitivity but something with how the overall total is calculated?

stackner, 2023-08-07:
[https://pasteboard.co/njJhAWs5lBJi.jpg](url)

[https://pasteboard.co/5K5zSDfmNBuA.jpg](url)

as you can see the daily shows 6kwh but the overall total shows for the same period 120kwh?

klaasnicolaas, 2023-08-07:
I see, I don't really have a solution for that right now. Maybe re-flashing will help? Can you share your yaml config?

closed 2023-09-13 by github-actions[bot]