klaasnicolaas/home-assistant-glow #299: web server frontpage is not displaying any data

poulanker, 2023-07-31:
I normally use home assistant to configure Glow and to present the data.
My implementation of Glow is using the default configuration, and is working well inside home assistant.
 
However, in the readme file I found the following advice for configuring e.g. the pulse rate:
  _"Open the webserver of the Glow and change the value under Pulse rate - imp/kWh."_

When I open the webserver and login, the webpage looks like this:
[image]

I can switch between a light and dark scheme, but that's all.
I have not been able to find a description or a picture showing the webpage as it should look.

Am I missing something in the configuration or are some files missing?
 
BR Poul Anker

klaasnicolaas, 2023-07-31:
Seems like your whole config is no longer present on your glow, you may have to reinstall the firmware to get everything back.

poulanker, 2023-08-03:
I have now installed Glow on a second device.
The served page looks similar, allthough this time the question marks are gone and the correct images are shown. 
I dived into the documentation of the webserver and found, that I could change the webserver version from 2 (default) to 1. Now the webserver serves a simple table, where I indeed can change e.g. the pulse rate.
How is the page supposed to look in the default version?

klaasnicolaas, 2023-08-04:
The table and log view side by side instead of on top of each other

closed 2023-09-10 by github-actions[bot]