klaasnicolaas/home-assistant-glow #300: Very strange readings after first setting up, why use GPIO 13?

WayneStarkey, 2023-08-01:
After I initially flashed the ESP32 and tested it before installing it on the meter I would get large readings very suddenly 

At first, I couldn't understand what was happening but then I realised it was when I touched the wiring to the Photodiode

Strange and with some head scratching! it dawned on me that GPIO13 is a touch pin,
and after reconfiguring it to GPIO26 it works perfectly and has been collecting data for over a week now
I've even tested it against the smart display and its within 2 watts every time  

[image]

The issue is why to use pin 13? has anyone else had this issue or is it because of the type of board I am using?

[](https://lastminuteengineers.com/esp32-pinout-reference/)

klaasnicolaas, 2023-08-02:
Earlier we used GPIO 12 and that caused problems for users, that's why it was changed to GPIO 13. So far I haven't heard any problems with that

WayneStarkey, 2023-08-03:
I also tried GPIO 12 and had the same problem as that is also a touch pin 
The reason for using GPIO 26 is it's not and it's working perfectly

klaasnicolaas, 2023-08-03:
https://lastminuteengineers.com/esp32-pinout-reference

According to this website GPIO 13 should be fine to use, which board do you have exactly?

WayneStarkey, 2023-08-03:
> https://lastminuteengineers.com/esp32-pinout-reference
> 
> According to this website GPIO 13 should be fine to use, which board do you have exactly?

The one you listed and thats the same on in the link above 

I did try the other touch pins and had the same issue and I also have another type of ESP32 (NodeMCU) board I tested with the same problem, I will test it again later and share the results

Looking though the other raised issues I feel others may have had a similar problems caused by the touch pin

klaasnicolaas, 2023-08-03:
Can you also try GPIO 5

closed 2023-09-25 by github-actions[bot]