klaasnicolaas/home-assistant-glow #301: Anyone else having issues with the upgrade to ESPhome 2023.8.1?

ABredhauer, 2023-08-18:
I installed the update and reflashed my glow but then it dropped off my network. Connected via USB and saw that it continually rebooted and died. For the moment I've rolled back to ESPHome 2023.7.1 and reflashed to get it working again

sillami1, 2023-08-18:
Installation went OK and device started OK. But now i get weird results:
Know that consumption ~242W is correct and ~3737W is wrong. This started after update...
Any idea what might cause this?

[09:35:52][D][light:047]:   State: ON
[09:35:52][D][sensor:094]: 'House - Power Consumption': Sending state 242.39127 W with 0 decimals of accuracy
[09:35:52][D][sensor:094]: 'House - Daily Energy': Sending state 1.55879 kWh with 3 decimals of accuracy
[09:35:52][D][sensor:094]: 'House - Total Energy': Sending state 0.32980 kWh with 3 decimals of accuracy
[09:35:52][D][light:036]: 'Red' Setting:
[09:35:52][D][sensor:094]: 'House - Power Consumption': Sending state 3737.54150 W with 0 decimals of accuracy
[09:35:52][D][sensor:094]: 'House - Daily Energy': Sending state 1.55888 kWh with 3 decimals of accuracy
[09:35:53][D][light:036]: 'Red' Setting:
[09:35:53][D][light:047]:   State: OFF
[09:35:53][D][light:036]: 'Red' Setting:
[09:35:54][D][sensor:094]: 'House - Total Energy': Sending state 0.32990 kWh with 3 decimals of accuracy
[09:35:54][D][light:036]: 'Red' Setting:
[09:35:54][D][light:047]:   State: ON
[09:35:54][D][sensor:094]: 'House - Power Consumption': Sending state 241.89825 W with 0 decimals of accuracy
[09:35:54][D][sensor:094]: 'House - Daily Energy': Sending state 1.55898 kWh with 3 decimals of accuracy
[09:35:54][D][sensor:094]: 'House - Total Energy': Sending state 0.33000 kWh with 3 decimals of accuracy
[09:35:54][D][light:036]: 'Red' Setting:
[09:35:54][D][sensor:094]: 'House - Power Consumption': Sending state 3737.26978 W with 0 decimals of accuracy
[09:35:54][D][sensor:094]: 'House - Daily Energy': Sending state 1.55908 kWh with 3 decimals of accuracy
[09:35:54][D][light:036]: 'Red' Setting:
[09:35:54][D][light:047]:   State: OFF
[09:35:54][D][light:036]: 'Red' Setting:
[09:35:56][D][sensor:094]: 'House - Total Energy': Sending state 0.33010 kWh with 3 decimals of accuracy
[09:35:56][D][light:036]: 'Red' Setting:

jerryzou, 2023-08-18:
> Installation went OK and device started OK. But now i get weird results: Know that consumption ~242W is correct and ~3737W is wrong. This started after update... Any idea what might cause this? [...]

Exactly the same thing happened at my end. I think the update broke the current configuration...

sillami1, 2023-08-18:
> Just noticed there is commented line in the configuration #internal_filter: 100ms I've uncommented it and changed the value to 200ms and reinstalled the fw. It seemed the issue went away. You could give it a go :)

That doesn't work correctly. If you have low consumption seems that it works but if I raise consumption eg. 4 kWh, the meter shows only ~half of that.

JuMaPa, 2023-08-18:
I have exact same problem.

klaasnicolaas, 2023-08-18:
I also updated my Glow to the latest version, but it works fine for me 🤷🏻‍♂️

klaasnicolaas, 2023-08-18:
@sillami1 What I see in your log is that it actually measures 1 pulse as 2 pulses because you see 2 times a watt log with the same timestamp 🤔

sillami1, 2023-08-18:
@klaasnicolaas Think the timestamp is not same, but fits inside same second. Here's other example...

[14:31:31][D][sensor:094]: 'House - Power Consumption': Sending state 309.58020 W with 0 decimals of accuracy
[14:31:32][D][sensor:094]: 'House - Daily Energy': Sending state 3.97142 kWh with 3 decimals of accuracy
[14:31:32][D][sensor:094]: 'House - Total Energy': Sending state 0.60900 kWh with 3 decimals of accuracy
[14:31:32][D][sensor:094]: 'House - Power Consumption': Sending state 3737.07617 W with 0 decimals of accuracy
[14:31:32][D][sensor:094]: 'House - Daily Energy': Sending state 3.97153 kWh with 3 decimals of accuracy
[14:31:33][D][sensor:094]: 'House - Total Energy': Sending state 0.60910 kWh with 3 decimals of accuracy

I'm not sure but seems it sends pulse of both, rising edge and falling edge. If so, the pulse of falling edge is incorrect.

JuMaPa, 2023-08-18:
Here is my log...

[14:59:34] [D] [sensor: 094]: 'House - Power Consumption': Sending state 3559.57739 W with 0 decimals of accuracy
[14:59:34] [D] [sensor: 094]: 'House - Daily
Energy': Sending state 22.30038 kWh with 3 decimals of accuracy
[14:59:34] [D] [sensor: 094]: 'House - Total
Energy': Sending state 3.16900 kWh with 3 decimals of accuracy
[14:59:34] [D] [sensor: 094]: 'House - Power Consumption': Sending state 117650.90625 W with 0 decimals of accuracy
[14:59:34] [D] [sensor: 094]: 'House - Daily Energy': Sending state 22.30215 kWh with 3 decimals of accuracy
[14:59:35] [D] [sensor: 094]: 'House - Total
Energy': Sending state 3.17000 kWh with 3 decimals of accuracy
[14:59:35] [D] [sensor: 094]: 'House - Power Consumption': Sending state 3545.55444 W with 0 decimals of accuracy
[14:59:35] [D] [sensor: 094]: 'House - Daily
Energy': Sending state 22.30312 kWh with 3 decimals of accuracy
[14:59:35] [D] [sensor: 094]: 'House - Total
Energy': Sending state 3.17100 kWh with 3 decimals of accuracy

jerryzou, 2023-08-18:
> That doesn't work correctly. If you have low consumption seems that it works but if I raise consumption eg. 4 kWh, the meter shows only ~half of that.

Tried with dishwasher and kettle on. The power went up to 4.5kw. Looks quite accurate. Maybe our setup is different.

JuMaPa, 2023-08-18:
In my case it seemed that min value show right value.
What can cause that and how i fix it?
Can someone help?

ABredhauer, 2023-08-18:
Seems that there's a few issues going on with the upgrade. The one that I had that I logged this issue for was

[web_server:116]: Setting up web server...
[C][ota:093]: Over-The-Air Updates:
[C][ota:094]:   Address: 192.168.1.59:3232
[C][api:025]: Setting up Home Assistant API server...
[C][mqtt:037]: Setting up MQTT...
[I][mqtt:246]: Connecting to MQTT...
[I][mqtt:286]: MQTT Connected!

abort() was called at PC 0x4020ce8b on core 0

Backtrace:0x40083985:0x3ffd80500x40093ec5:0x3ffd8070 0x40099489:0x3ffd8090 0x4020ce8b:0x3ffd8110 0x4020ced2:0x3ffd8130 0x4020cfcb:0x3ffd8150 0x4020cf2a:0x3ffd8170 0x400dcc89:0x3ffd8190 0x4012e555:0x3ffd81b0 0x401576d9:0x3ffd81d0 0x40159533:0x3ffd81f0

klaasnicolaas, 2023-08-19:
Please add a full configuration file you are using with the secrets redacted.

JuMaPa, 2023-08-19:
> Tried with dishwasher and kettle on. The power went up to 4.5kw. Looks quite accurate. Maybe our setup is different.

This look's like fix my problem also.

sillami1, 2023-08-19:
@JuMaPa What's your IMP value?

JuMaPa, 2023-08-19:
> @JuMaPa What's your IMP value? 

Not sure what u mean, but 1000imp/kwh

JuMaPa, 2023-08-19:
And just check power consumption history graph...it's not right even now.

sillami1, 2023-08-19:
Yes, just that what I meant. I have 10000 so I have 10 times more pulses. And that's why the mentioned solution don't work for me. There are also reported this issue in ESPhome GitHub https://github.com/esphome/issues/issues/4807

fabianschaefer, 2023-08-20:
I also have another issue with pulse meter since the [refactor](https://github.com/esphome/esphome/pull/4231). The "no pulse detected" code now always ignores the last pulse. In my case I have the timeout set to 10s. I get a pulse at 01:44:05 and another at 01:44:09, but at 01:44:15 (10 seconds after the penultimate pulse), I get the "No pulse detected for 10s message" and the sensor goes to 0, which is wrong:
```[01:44:05][D][sensor:094]: 'water_meter_flow': Sending state 18.74292 L/min with 0 decimals of accuracy
[01:44:09][D][pulse_meter:057]: No pulse detected for 10s, assuming 0 pulses/min
[01:44:09][D][sensor:094]: 'water_meter_flow': Sending state 0.00000 L/min with 0 decimals of accuracy
[01:44:11][D][sensor:094]: 'water_meter_total': Sending state 38.00000 L with 0 decimals of accuracy
[01:44:15][D][pulse_meter:057]: No pulse detected for 10s, assuming 0 pulses/min
[01:44:15][D][sensor:094]: 'water_meter_flow': Sending state 0.00000 L/min with 0 decimals of accuracy
[01:44:17][D][sensor:094]: 'water_meter_total': Sending state 39.00000 L with 0 decimals of accuracy
[01:44:21][D][pulse_meter:057]: No pulse detected for 10s, assuming 0 pulses/min
[01:44:21][D][sensor:094]: 'water_meter_flow': Sending state 0.00000 L/min with 0 decimals of accuracy
[01:44:23][D][sensor:094]: 'water_meter_total': Sending state 40.00000 L with 0 decimals of accuracy
[01:44:27][D][pulse_meter:057]: No pulse detected for 10s, assuming 0 pulses/min```

klaasnicolaas, 2023-08-20:
@sillami1 10.000 pulses is actually too much to use the pulse_meter component properly, which may also explain why your measurements are not going well.

pikkmarg, 2023-08-20:
I've just updated to 2018.8.1 today and have been facing the same issues. Multiple reportings at the same time and creating values well over 200,000,000 at some points.

Changing the internal filter to 100ms did work in getting the values to be more realistic and removed the jumpy results from the multiple pulses at the same time. I'm not sure however why the sensitivity of the sensor took a leap after the update.

ABredhauer, 2023-08-21:
@klaasnicolaas thanks for following up. I have not made any progress but I've also come down with the flu so haven't done much of anything
I have uploaded my config here https://pastebin.com/LaRB54tt
and a bit more of the log here https://pastebin.com/41jYrHUy

It seems to always die after MQTT so I'm wondering if thats the issue

For now I rolled back my ESPHome to the previous release and reflashed the firmware.

klaasnicolaas, 2023-08-21:
Your pastebins are private

ABredhauer, 2023-08-21:
Sorry should be fixed now

klaasnicolaas, 2023-08-21:
Would start by disabling the MQTT (why are you using that?) and the BLE proxy, see if that makes a difference.

ABredhauer, 2023-08-21:
Actually I’m not using the mqtt anymore. It was a workaround when I was having connectivity issues with my wifi and for some reason the connection was much more stable with mqtt enabled. I’ll disable and check it in the morning. And yes the proxy upgrade was done via cable initially. It’s been stable for months with no issues until this update

jysaloma, 2023-08-21:
I am having this same issue as well. Has been working fine for a year.

ABredhauer, 2023-08-22:
@klaasnicolaas I removed the MQTT setting from my config and reflashed while staying on 2023.7 and my glow and connection were stable for an hour so I bit the bullet and upgraded ESPHome to 2023.8.2 and reflashed with updated config. It still seems to be stable after 4 hours so I guess it was the MQTT after all. Thanks for your help
FYI I'm not seeing the pulse meter issue that everyone else has been reporting

pikkmarg, 2023-08-22:
@JuMaPa Are you seeing multiple inputs when the red light flashes? 

In this case uncommenting `#internal_filter: 100ms` might get you closer since it will filter the responses and limit input to 100ms. Someone had luck with 200ms.

Very high readings usually pop up for me when the sensitivity of the sensor changes for some reason. I then usually manually calibrate the sensor. This is the first time i´ve used the filter and it seems to be a good combination to get the readings clear.

JuMaPa, 2023-08-22:
> @JuMaPa Are you seeing multiple inputs when the red light flashes? 
> 
> 
> 
> In this case uncommenting `#internal_filter: 100ms` might get you closer since it will filter the responses and limit input to 100ms. Someone had luck with 200ms but 100 worked perfectly for me. 
> 
> 
> 
> Very high readings usually pop up for me when the sensitivity of the sensor changes for some reason. I then usually manually calibrate the sensor. This is the first time i´ve used the filter and it seems to be a good combination to get the readings clear. 

> Here is my log...
>
> [14:59:34] [D] [sensor: 094]: 'House - Power Consumption': Sending state 3559.57739 W with 0 decimals of accuracy
> [14:59:34] [D] [sensor: 094]: 'House - Daily
> Energy': Sending [...]

That's my log when my problem occured.

I try to change after that above log 200ms, it is closer. But history log looks like pillar graph after that.

Hope undersstand what i try to explain.

pikkmarg, 2023-08-23:
@JuMaPa Yeah you have to use the blue potentiometer on the sensor so that you get as few readings when the red light flashes as possible. Then the filter works.

Just to clarify further.
Step 1: Calibrate the sensor manually so that it would be as close as you can to 1 reading per red flash.
Step 2: Fine tune with the filter if the manual calibration is not sensitive enough and you have 2-3 readings per red flash.

Hope this made sense.

dartfrogdk, 2023-08-27:
same issue updated today to 2023.8.2, too high readings, been 100% stable for years, something changed

pikkmarg, 2023-08-28:
> same issue updated today to 2023.8.2, too high readings, been 100% stable for years, something changed

Did you try recalibrating the sensor?

ABredhauer, 2023-08-28:
So the issue I had with the upgrade was different and I didnt see the strange readings that others are seeing but I did have a similar issue much earlier in my journey and solved it with some of the info in this issue https://github.com/klaasnicolaas/home-assistant-glow/issues/57 

Wondering if thats why I'm not seeing this issue now

pikkmarg, 2023-08-28:
> So the issue I had with the upgrade was different and I didnt see the strange readings that others are seeing but I did have a similar issue much earlier in my journey and solved it with some of the info in this issue #57
> 
> Wondering if thats why I'm not seeing this issue now

For clarity then if you have you issue solved then please mark the topic as anwsered and we can continue in #302 since the nature of the issues are actually different and you topic has been hijacked for a new cause. 

Thanks!

ABredhauer, 2023-08-28:
Sounds good to me

closed 2023-08-28 by ABredhauer

dartfrogdk, 2023-08-28:
like i just said it have to do with the update, behaviour changed right
after, sensor is where it should be, i have upgraded every update without
ANY problems