klaasnicolaas/home-assistant-glow #302: Power consumption over the roof

Proche72, 2023-08-27:
Hey, on 17th of August around 9pm my HA Glow went haywire and start measuring 183 kW power consumption alp day (my daily over some is ~12 kWh). 
[image]

Since then my Energy board in HA is completely wrong. From 0.31 kWh I have 0.64 kWh base consumption. From ~12 kWh a day I have 23 kWh. Anyone seen this behaviour?

INFO ESPHome 2023.8.2
INFO Reading configuration /config/esphome/home-assistant-glow.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Prague'
INFO Starting log output from home-assistant-glow.local using esphome API
INFO Successfully connected to home-assistant-glow.local
[19:33:48][I][app:102]: ESPHome version 2023.8.2 compiled on Aug 21 2023, 21:03:02
[19:33:48][I][app:104]: Project klaasnicolaas.home-assistant-glow version 3.1.0
[19:33:48][C][status_led:019]: Status LED:
[19:33:48][C][status_led:020]:   Pin: GPIO5
[19:33:48][C][wifi:543]: WiFi:
[19:33:48][C][wifi:379]:   Local MAC: 08:3A:F2:59:A0:C4
[19:33:48][C][wifi:380]:   SSID: [redacted]
[19:33:48][C][wifi:381]:   IP Address: 192.168.0.143
[19:33:48][C][wifi:383]:   BSSID: [redacted]
[19:33:48][C][wifi:384]:   Hostname: 'home-assistant-glow'
[19:33:48][C][wifi:386]:   Signal strength: -75 dB ▂▄▆█
[19:33:48][C][wifi:390]:   Channel: 2
[19:33:48][C][wifi:391]:   Subnet: 255.255.255.0
[19:33:48][C][wifi:392]:   Gateway: 192.168.0.1
[19:33:48][C][wifi:393]:   DNS1: 8.8.8.8
[19:33:48][C][wifi:394]:   DNS2: 8.8.4.4
[19:33:48][C][logger:301]: Logger:
[19:33:48][C][logger:302]:   Level: DEBUG
[19:33:48][C][logger:303]:   Log Baud Rate: 115200
[19:33:48][C][logger:305]:   Hardware UART: UART0
[19:33:48][C][template.number:050]: Template Number 'Puls rate - imp/kWh'
[19:33:48][C][template.number:051]:   Optimistic: YES
[19:33:48][C][template.number:052]:   Update Interval: 60.0s
[19:33:48][C][gpio.output:010]: GPIO Binary Output:
[19:33:48][C][gpio.output:011]:   Pin: GPIO2
[19:33:48][C][gpio.output:010]: GPIO Binary Output:
[19:33:48][C][gpio.output:011]:   Pin: GPIO4
[19:33:48][C][template.text_sensor:020]: Template Sensor 'Glow - Installed version'
[19:33:48][C][template.text_sensor:020]:   Icon: 'mdi:label-outline'
[19:33:48][C][light:103]: Light 'Red'
[19:33:48][C][restart.button:017]: Restart Button 'Restart - Glow'
[19:33:48][C][version.text_sensor:021]: Version Text Sensor 'House - ESPHome Version'
[19:33:48][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[19:33:48][C][pulse_meter:067]: Pulse Meter 'House - Power Consumption'
[19:33:48][C][pulse_meter:067]:   Device Class: 'power'
[19:33:48][C][pulse_meter:067]:   State Class: 'measurement'
[19:33:48][C][pulse_meter:067]:   Unit of Measurement: 'W'
[19:33:48][C][pulse_meter:067]:   Accuracy Decimals: 0
[19:33:48][C][pulse_meter:067]:   Icon: 'mdi:flash-outline'
[19:33:48][C][pulse_meter:068]:   Pin: GPIO13
[19:33:48][C][pulse_meter:070]:   Filtering rising edges less than 13 µs apart
[19:33:48][C][pulse_meter:074]:   Assuming 0 pulses/min after not receiving a pulse for 300s
[19:33:48][C][total_daily_energy:023]: Total Daily Energy 'House - Daily Energy'
[19:33:48][C][total_daily_energy:023]:   Device Class: 'energy'
[19:33:48][C][total_daily_energy:023]:   State Class: 'total_increasing'
[19:33:48][C][total_daily_energy:023]:   Unit of Measurement: 'kWh'
[19:33:48][C][total_daily_energy:023]:   Accuracy Decimals: 3
[19:33:48][C][total_daily_energy:023]:   Icon: 'mdi:circle-slice-3'
[19:33:48][C][homeassistant.time:010]: Home Assistant Time:
[19:33:48][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[19:33:48][C][captive_portal:088]: Captive Portal:
[19:33:48][C][web_server:161]: Web Server:
[19:33:48][C][web_server:162]:   Address: home-assistant-glow.local:80
[19:33:48][C][mdns:112]: mDNS:
[19:33:48][C][mdns:113]:   Hostname: home-assistant-glow
[19:33:48][C][ota:093]: Over-The-Air Updates:
[19:33:48][C][ota:094]:   Address: home-assistant-glow.local:3232
[19:33:48][C][api:138]: API Server:
[19:33:48][C][api:139]:   Address: home-assistant-glow.local:6053
[19:33:48][C][api:143]:   Using noise encryption: NO
[19:33:48][C][improv_serial:032]: Improv Serial:
[19:33:48][C][wifi_info:009]: WifiInfo IPAddress 'House - IP Address'
[19:33:48][C][wifi_info:009]:   Icon: 'mdi:wifi'
[19:33:48][C][wifi_info:011]: WifiInfo SSID 'House - Connected SSID'
[19:33:48][C][wifi_info:011]:   Icon: 'mdi:wifi-strength-2'
[19:33:48][C][wifi_signal.sensor:009]: WiFi Signal 'House - WiFi Signal'
[19:33:48][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[19:33:48][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[19:33:48][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[19:33:48][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[19:33:48][D][esp32.preferences:114]: Saving 1 preferences to flash...
[19:33:48][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[19:33:59][D][sensor:094]: 'House - Total Energy': Sending state 24.20400 kWh with 3 decimals of accuracy
[19:33:59][D][sensor:094]: 'House - Power Consumption': Sending state 401.73999 W with 0 decimals of accuracy
[19:33:59][D][light:036]: 'Red' Setting:
[19:33:59][D][light:047]:   State: ON
[19:33:59][D][sensor:094]: 'House - Daily Energy': Sending state 27.37893 kWh with 3 decimals of accuracy
[19:33:59][D][sensor:094]: 'House - Total Energy': Sending state 24.20600 kWh with 3 decimals of accuracy
[19:33:59][D][sensor:094]: 'House - Power Consumption': Sending state 183043.09375 W with 0 decimals of accuracy
[19:33:59][D][light:036]: 'Red' Setting:
[19:33:59][D][sensor:094]: 'House - Daily Energy': Sending state 27.38229 kWh with 3 decimals of accuracy
[19:34:00][D][light:036]: 'Red' Setting:
[19:34:00][D][light:047]:   State: OFF
[19:34:00][D][light:036]: 'Red' Setting:

dartfrogdk, 2023-08-27:
got the same problem

klaasnicolaas, 2023-08-27:
Duplicate of #301?

pikkmarg, 2023-08-28:
**Did you try recalibrating the sensor to fix the issue?**
Sometimes the big updates mess with the calibration but often times than not, for me, the two-sided tape that is holding the sensor on the panel starts melting and the sensor slips out of place.

I have an off-white enclosure and when the red light hits the sensor in an angle then the readings can get messy. I just had my sensor go haywire at 5 am this morning starting with 8000w readings and getting to 300,000 in a few hours with correct readings in between. 

[image]

Since the sensor needs to be quite sensitive for the red light to trigger it then any bouncing light seems to make it read it twice. As has been suggested in previous issues then recalibrating and, if needed, a further filter implemented to get correct readings from the panel.

Proche72, 2023-08-28:
> Duplicate of #301?

To be honest...I don't know. I don't know when I upgraded to 2023.8.x but yeah, could be. I will go over the the #301 again to try understand it. But from what I get is more ESPHome issue then the HA Glow.

Proche72, 2023-08-28:
> **Did you try recalibrating the sensor to fix the issue?** Sometimes the big updates mess with the calibration but often times than not, for me, the two-sided tape that is holding the sensor on the panel starts melting and the sensor slips out of place. I have another piece of tape going accross but it seems that the outer case is held in place but it slowly slides underneath where the enclosure meets.
> 
> I have an off-white enclosure and when the red light hits the sensor at an angle then the readings can get messy. I just had my sensor go haywire at 5 am this morning starting with 8000w readings and getting to 300,000 in a few hours with correct readings in between.
> 
> Since the sensor needs to be quite sensitive for the red light to trigger it then any bouncing light seems to make it read it twice. As has been suggested in previous issues then recalibrating and, if needed, a further filter implemented to get correct readings from the panel.
> 
> In case this is important - HASS 2023.8.4 and ESPhome 2023.8.2

I did not. I will try reposition the sensor over the LED. I have checked it visually and it seemed fine - green led on electrometer flashed and immediatelly the red one on the HA Glow flashed too. So I reckoned that the measuring is working. But getting the flashing twice as it move slightly down could be the problem. 

Is there any more recalibration possible? Or just repositioning?

Thank you!

pikkmarg, 2023-09-01:
@Proche72 Seems like its been running smooth ever since? If so then please close the topic with a positive resolution :)
Have a happy friday!

dartfrogdk, 2023-09-01:
its not acurate when using the delay, so it is not fixed, why wont you see that many people have this problem it is a software change somewhere

klaasnicolaas, 2023-09-01:
Then you should rather raise this with ESPHome and not here, if it is a software problem 😉

Proche72, 2023-09-01:
> @Proche72 Seems like its been running smooth ever since? If so then please close the topic with a positive resolution :) Have a happy friday!

Sorry, I was nit at home until today.

Repositioning and recalibrating (I chanfe the settings on it) helped. Now I am getting normal values in consumption (500W). Hopefully it is correct.

Proche72, 2023-09-01:
Just recap with solution:

Repositioning - the adhesive I used was foamy and it moved bit down on the foam.

Recalibrating - change the settings on the dial by one lower

closed 2023-09-01 by Proche72