klaasnicolaas/home-assistant-glow #309: No reading

dean-ElisGroup, 2023-10-04:
Anyone experienced no reading since the ESPHome updates recently?

My Glow does not read anything now and shows just a ZERO for usage.  i've tried resetting etc but nothing is working.

klaasnicolaas, 2023-10-04:
Which version are you using and is your sensor still on the correct GPIO pin?

sunishnet, 2023-10-05:
I too have noticed recently, that once in a while its stopping metering. Was thinking it had something to do with my wifi network.

pikkmarg, 2023-10-05:
I updated the position of a wifi repeater that is now used by the ESP and had to recalibrate the sensor.

Not sure but maybe it has to do with some kind of change in the amount of power going through the module?

dean-ElisGroup, 2023-10-05:
> I updated the position of a wifi repeater that is now used by the ESP and had to recalibrate the sensor.
> 
> Not sure but maybe it has to do with some kind of change in the amount of power going through the module?
> 
> This might relate to the ESPHome updates causing some optimisations to mess with the power levels and thus causing the sensor to become less reliable.

I've changed nothing, other than the software update.  How do you recalibrate the sensor?

pikkmarg, 2023-10-05:
> I've changed nothing, other than the software update. How do you recalibrate the sensor?

The sensor has a blue potentiometer that, when turned, changes the sensitivity of the sensor.

dean-ElisGroup, 2023-10-09:
I get sporadic readings now, a few minutes ago it was over 11,000,000W, not sure that's correct!

pikkmarg, 2023-10-09:
Yes, if you look at the log then this happens because the sensitivity is such that it reports the same red flash as multiple flashes. 

There are two ways to correct this:

- Turn the potentiometer so that it only provides only one red flash
- Use the internal filter inside YAML by uncommenting it. `#internal_filter: 100ms`

Barma-lej, 2023-11-05:
Hi guys. In ESPHome 2023.8.0, there was a pulse_meter refactoring. The default mode is now EDGE instead of PULSE as before.

To use the PULSE mode, you need to add the internal_filter_mode option to the pulse_meter configuration, like this:

```
  - platform: pulse_meter
    name: '${friendly_name} - Power Consumption'
    id: sensor_energy_pulse_meter
    internal_filter_mode: PULSE
    unit_of_measurement: 'W'
    state_class: measurement
    ...
```

klaasnicolaas, 2023-11-06:
I use the EDGE mode with an `internal_filter` of 100ms, with a pulse rate of 1000 and have not noticed any difference 🤔

Barma-lej, 2023-11-06:
Do you have any strangers values?

klaasnicolaas, 2023-11-06:
Nope, it is always spot on with measuring

Barma-lej, 2023-11-06:
hm, In EDGE mode, an `internal_filter` of 100ms and a pulse rate of 10000 i have right values too 🤔

dean-ElisGroup, 2023-11-06:
Mine is still dead. Reports a 0w all day every day.  I've tried to fix it but with no luck

klaasnicolaas, 2023-11-07:
A pulse rate of 10000 is actually too high to make correct measurements. But if I understand correctly, you get good measurements when you set the `interval_filter_mode` to PULSE?

klaasnicolaas, 2023-11-07:
Perhaps it would be better to set the `internal_filter` with 100ms default and not the mode back to PULSE as in your PR.

Barma-lej, 2023-11-07:
I did close it

closed 2023-12-16 by github-actions[bot]