klaasnicolaas/home-assistant-glow #315: Firmware stuck on v2023.2.4

cieska, 2023-11-08:
Hi,

I have no experience with ESP32, but I'm just smart enough to make it work with home-assistant-glow. And it was running fine not counting the fact that on my system it stuck with the firmware version 2023.2.4 and it is not upgrading any more as it is now. I'm upgrading the home assistant occasionally, and other components, but only have an issue with HA-glow. 
Any ideas? 
[image]

cieska, 2023-11-08:
Just today I updated the config, as I though it may change anything, but no.

klaasnicolaas, 2023-11-09:
Try updating inside the ESPHome add-on, then you would also see the build logs. Perhaps it contains interesting information about why things go wrong.

cieska, 2023-11-09:
I'm a bit unsure how this must be done, I mean updating inside the ESPHome add-on. But I guess I go to ESPhome >> 3dot menu >> Install >> Choose the way to connect/install to the device.

So far I have got the first error 
```
Updating /config/esphome/home-assistant-glow.yaml
------------------------------------------------------------

INFO ESPHome 2023.10.6
INFO Reading configuration /config/esphome/home-assistant-glow.yaml...
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/London'
INFO Generating C++ source...
INFO Compiling app...
Processing home-assistant-glow (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
|-- noise-c @ 0.1.4
Compiling .pioenvs/home-assistant-glow/src/main.cpp.o
Linking .pioenvs/home-assistant-glow/firmware.elf
RAM:   [=         ]  12.6% (used 41212 bytes from 327680 bytes)
Flash: [=====     ]  54.6% (used 1002653 bytes from 1835008 bytes)
Building .pioenvs/home-assistant-glow/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/home-assistant-glow/firmware.bin"], [".pioenvs/home-assistant-glow/firmware.elf"])
Wrote 0x106330 bytes to file /data/build/home-assistant-glow/.pioenvs/home-assistant-glow/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 77.42 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of home-assistant-glow.local
INFO  -> 192.168.0.64
INFO Uploading /data/build/home-assistant-glow/.pioenvs/home-assistant-glow/firmware.bin (1008432 bytes)
ERROR ESP requests password, but no password given!
===== [ERROR] /config/esphome/home-assistant-glow.yaml =====

======================== [SUMMARY] ========================
  - /config/esphome/home-assistant-glow.yaml: FAILED
```
Then after chasing YAML errors, I'm stuck with having short some sort of credentials

```
Updating /config/esphome/home-assistant-glow.yaml
------------------------------------------------------------

INFO ESPHome 2023.10.6
INFO Reading configuration /config/esphome/home-assistant-glow.yaml...
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/London'
INFO Generating C++ source...
INFO Compiling app...
Processing home-assistant-glow (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
|-- noise-c @ 0.1.4
Compiling .pioenvs/home-assistant-glow/src/main.cpp.o
Linking .pioenvs/home-assistant-glow/firmware.elf
RAM:   [=         ]  12.6% (used 41212 bytes from 327680 bytes)
Flash: [=====     ]  54.6% (used 1002653 bytes from 1835008 bytes)
Building .pioenvs/home-assistant-glow/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/home-assistant-glow/firmware.bin"], [".pioenvs/home-assistant-glow/firmware.elf"])
Wrote 0x106330 bytes to file /data/build/home-assistant-glow/.pioenvs/home-assistant-glow/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 77.42 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of home-assistant-glow.local
INFO  -> 192.168.0.64
INFO Uploading /data/build/home-assistant-glow/.pioenvs/home-assistant-glow/firmware.bin (1008432 bytes)
ERROR ESP requests password, but no password given!
===== [ERROR] /config/esphome/home-assistant-glow.yaml =====

======================== [SUMMARY] ========================
  - /config/esphome/home-assistant-glow.yaml: FAILED
```
What kind of passwords?

klaasnicolaas, 2023-11-15:
These refer to your secrets file with all your passwords for ESPHome, you can also remove this and replace it with something yourself

closed 2023-12-24 by github-actions[bot]