klaasnicolaas/home-assistant-glow #319: House Total Energy get reset

peros550, 2024-01-01:
Many thanks for your work. 
I have a small issue and thought I should report it. The House Total Energy sensor gets back to zero every few minutes. 

I can set it again to sync it with my home's meter but it resets by itself every few minutes. I am using an esp32 (nodemcu 30 pin) board with the following specification

  # Define the GPIO pins
  pulse_pin: GPIO23
  status_led: GPIO18
  led_pin_red: GPIO02
  led_pin_green: GPIO19

klaasnicolaas, 2024-01-01:
Could it be that your ESP is constantly restarting or that it often loses the connection?

peros550, 2024-01-02:
Thanks for the quick feedback. ESP's wifi signal seems strong, just a few meters from the AP and no walls but I don't know if it reboots itself. 

So if this to be expected when the connection to the HA is not rock solid or the esp looses power, that's ok.  On the other hand, the house daily energy seems to work nicely and does not get reset. Only the house total power gets reset.

closed 2024-02-10 by github-actions[bot]

tomaszrondio, 2024-09-13:
have you found a solution? I get total energy reset every home assistant reboot. Turbo annoying

klaasnicolaas, 2024-09-13:
It is a default behavior that the total entity always resets after reboot, there is a [PR open](https://github.com/klaasnicolaas/home-assistant-glow/pull/527) to make it a persistent value. Just haven't had the time to test it myself yet.

./Klaas

tomaszrondio, 2024-09-14:
I think I solved it by using HA utility meter. Will see how convenient is that