klaasnicolaas/home-assistant-glow #358: Error downloading manifest

Adhd2012, 2024-01-13:
When I try to install Glow, I only get a small window that says Error! Failed to download manifest. Have tried on several different devices, both ESP8266 and ESP32WROOM and always the same error message. What am I doing wrong? How can I fix this?
[image]

klaasnicolaas, 2024-01-13:
It is possible that the ESP32WROOM does not work because it is a different board type, but the ESP8266 should work. This is something I need to investigate further.

Which boards do you have exactly? Link?

Adhd2012, 2024-01-13:
just a normal esp8266mod that i bought on aliexpress

klaasnicolaas, 2024-01-14:
The only thing I can think of is that you have an ESP with a board type that differs from the built firmwares. Just tested it with my own ESP32S and it went well. I may need to change the board types to something more general 🤔 

Could you please send pictures of your board? where chip and board type are clearly readable.

klaasnicolaas, 2024-01-14:
What you also could do is open the developer tools (short key: F12) of Chrome during installation and then on the console tab. This provides interesting information if you get that manifest error and helps me with debugging.

Adhd2012, 2024-01-15:
Yes, ok, I'll try it. Hope these pictures are good enough. If not, let me know.
[image]
[image]

Adhd2012, 2024-01-15:
I'm very sorry about this issue because today when I was going to do what you recommended regarding f12 for troubleshooting everything went perfectly fine. No error message now. Very strange, but super good for me :-). Btw I installed glow on esp8266, same device that didn't work yesterday.

Adhd2012, 2024-01-15:
That means that now I can't progress from step 2. Now it just wants to install home assistant glow over and over again every time I press next. Going to try what you talked about now regarding troubleshooting.

Adhd2012, 2024-01-15:
I just copied the log. Hope that's good enough, if not you'll have to tell me what to do.

ESP Web Tools 9.4.3 by Nabu Casa; https://esphome.github.io/esp-web-tools/
install-dialog-28H-HNrD.js?module:424 Initializing Improv Serial
install-dialog-28H-HNrD.js?module:724 Improv initialization failed. Error: Improv Wi-Fi Serial not detected
    at install-dialog-28H-HNrD.js?module:424:1355
_initialize @ install-dialog-28H-HNrD.js?module:724
install-dialog-28H-HNrD.js?module:424 esptool.js
install-dialog-28H-HNrD.js?module:424 Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
install-dialog-28H-HNrD.js?module:424 Connecting...
install-dialog-28H-HNrD.js?module:424 .
install-dialog-28H-HNrD.js?module:424 

install-dialog-28H-HNrD.js?module:424 Detecting chip type... 
install-dialog-28H-HNrD.js?module:424 ESP8266
install-dialog-28H-HNrD.js?module:424 Chip is ESP8266EX
install-dialog-28H-HNrD.js?module:424 Features: WiFi
install-dialog-28H-HNrD.js?module:424 Crystal is 26MHz
install-dialog-28H-HNrD.js?module:424 MAC: d8:f1:5b:14:27:4c
install-dialog-28H-HNrD.js?module:424 Uploading stub...
install-dialog-28H-HNrD.js?module:424 Running stub...
install-dialog-28H-HNrD.js?module:424 Stub running...
install-dialog-28H-HNrD.js?module:424 Manufacturer: 0
install-dialog-28H-HNrD.js?module:424 Device: 00
install-dialog-28H-HNrD.js?module:424 Detected flash size: undefined
install-dialog-28H-HNrD.js?module:424 Erasing flash (this may take a while)...
install-dialog-28H-HNrD.js?module:424 Chip erase completed successfully in 1.765s
install-dialog-28H-HNrD.js?module:424 Not changing the image
install-dialog-28H-HNrD.js?module:424 Compressed 494800 bytes to 339651...
install-dialog-28H-HNrD.js?module:424 Writing at 0x0... (4%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x62ff... (9%)
install-dialog-28H-HNrD.js?module:424 Writing at 0xc282... (14%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x12491... (19%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x187aa... (23%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x1eac4... (28%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x24758... (33%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x2a8f9... (38%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x31b85... (42%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x3780a... (47%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x3d01a... (52%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x42861... (57%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x47cd5... (61%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x4cc28... (66%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x51dcc... (71%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x57473... (76%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x5cb2e... (80%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x620e1... (85%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x6762e... (90%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x6e738... (95%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x7432f... (100%)
install-dialog-28H-HNrD.js?module:424 Wrote 494800 bytes (339651 compressed) at 0x0 in 31.259 seconds.
install-dialog-28H-HNrD.js?module:424 Leaving...
install-dialog-28H-HNrD.js?module:724 HARD RESET
install-dialog-28H-HNrD.js?module:724 DISCONNECT
install-dialog-28H-HNrD.js?module:424 Initializing Improv Serial
install-dialog-28H-HNrD.js?module:724 Improv initialization failed. Error: Improv Wi-Fi Serial not detected
    at install-dialog-28H-HNrD.js?module:424:1355
_initialize @ install-dialog-28H-HNrD.js?module:724
getting-started:1 Uncaught (in promise) Error: A listener indicated an asynchronous response by returning true, but the message channel closed before a response was received
install-dialog-28H-HNrD.js?module:424 esptool.js
install-dialog-28H-HNrD.js?module:424 Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
install-dialog-28H-HNrD.js?module:424 Connecting...
install-dialog-28H-HNrD.js?module:424 .
install-dialog-28H-HNrD.js?module:424 

install-dialog-28H-HNrD.js?module:424 Detecting chip type... 
install-dialog-28H-HNrD.js?module:424 ESP8266
install-dialog-28H-HNrD.js?module:424 Chip is ESP8266EX
install-dialog-28H-HNrD.js?module:424 Features: WiFi
install-dialog-28H-HNrD.js?module:424 Crystal is 26MHz
[... 15 lines cut ...]
install-dialog-28H-HNrD.js?module:424 Writing at 0x187aa... (23%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x1eac4... (28%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x24758... (33%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x2a8f9... (38%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x31b85... (42%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x3780a... (47%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x3d01a... (52%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x42861... (57%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x47cd5... (61%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x4cc28... (66%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x51dcc... (71%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x57473... (76%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x5cb2e... (80%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x620e1... (85%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x6762e... (90%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x6e738... (95%)
install-dialog-28H-HNrD.js?module:424 Writing at 0x7432f... (100%)
install-dialog-28H-HNrD.js?module:424 Wrote 494800 bytes (339651 compressed) at 0x0 in 31.163 seconds.
install-dialog-28H-HNrD.js?module:424 Leaving...
install-dialog-28H-HNrD.js?module:724 HARD RESET
install-dialog-28H-HNrD.js?module:724 DISCONNECT
install-dialog-28H-HNrD.js?module:424 Initializing Improv Serial
install-dialog-28H-HNrD.js?module:724 Improv initialization failed. Error: Improv Wi-Fi Serial not detected
    at install-dialog-28H-HNrD.js?module:424:1355
_initialize @ install-dialog-28H-HNrD.js?module:724

klaasnicolaas, 2024-01-15:
I've also made some adjustments myself in recent days, this may explain why it went well now. Many things with improv are also new to me, so as far as I can see it is going well (except installing over and over). Perhaps reconnecting will allow you to proceed to the step to enter the WiFi credentials.

klaasnicolaas, 2024-01-25:
Do you have an update yet on whether things are going well now?

closed 2024-02-04 by klaasnicolaas