klaasnicolaas/home-assistant-glow #426: Installation issue, No such file or directory when trying to install updated config

paulmck50, 2024-03-22:
I've installed the latest Glow package which installed onto my ESP32 (AZ-Delivery ESP32 Dev KitC V4), initial install works ok no errors and connects to Wifi Successfully, it then becomes available to adopt in HA which again works and changes status to online, but then I'm not able to update the configuration / install changes.  I get the below error when trying via serial or wifi update.

I've tried copying the files manually to the esphome folder but that has made no difference.  

INFO ESPHome 2024.3.0
INFO Reading configuration /config/esphome/home-assistant-glow-ccc660.yaml...
ERROR Error while reading config: Invalid YAML syntax:

Error reading file /config/esphome/../components/basis.yaml: [Errno 2] No such file or directory: '/config/esphome/../components/basis.yaml'

klaasnicolaas, 2024-03-22:
What does your glow config file look like? Because with the current firmware on the website, you would no longer have to do anything with the individual component files after adoption.

./klaas

klaasnicolaas, 2024-03-22:
I've a fix ready and I will also update the documentation later with instructions on how to customize the `pulse_meter` component.

./Klaas

PR #427 (2024-03-22): Fix issue with package import during build
This will fix the import issue after device adoption in ESPHome dashboard.

Closes: #426
closed 2024-03-22 by klaasnicolaas