klaasnicolaas/home-assistant-glow #452: Reduce amount of data from sensor_total_energy

Xilef00, 2024-04-06:
Hello, I recently got a Home Assistant Glow and it's working great.

To reduce the number of sensor updates, I added the following to my .yaml file in ESPHome as described in the documentation:

```
sensor:
  - id: !extend sensor_energy_pulse_meter
    internal_filter: 100ms
  - id: !extend sensor_energy_pulse_meter
    filters:
      - throttle_average: 10s
      - filter_out: NaN
  - id: !extend sensor_total_daily_energy
    filters:
      - delta: 0.01
      - heartbeat: 300s
```

Now the "Power Consumption" sensor only sends updates every 10 seconds, and the sensor_total_daily_energy follows the specified config. However, the sensor_total_energy still sends updates approximately every second:

```
[10:49:57][D][sensor:094]: 'Home Assistant Glow - Total Energy': Sending state 4.17900 kWh with 3 decimals of accuracy
[10:49:58][D][sensor:094]: 'Home Assistant Glow - Total Energy': Sending state 4.17910 kWh with 3 decimals of accuracy
[10:50:00][D][sensor:094]: 'Home Assistant Glow - Total Energy': Sending state 4.17920 kWh with 3 decimals of accuracy
```
I'd like to use the same filters for this sensor as with sensor_total_daily_energy. However, since it's configured within the platform as "pulse_meter", I'm unsure how to add an extension.

When I insert the sensor as shown below, I receive a message in the editor that the source for the extension could not be found.

```
sensor:
  - id: !extend sensor_energy_pulse_meter
    internal_filter: 100ms
  - id: !extend sensor_energy_pulse_meter
    filters:
      # Update the sensor with an average every 10th second. See
      - throttle_average: 10s
      - filter_out: NaN
  - id: !extend sensor_total_energy
    filters:
      # Update the sensor once per 0.1 kWh consumed, or every 5th minute, whichever happens sooner.
      - delta: 0.1
      - heartbeat: 300s
  - id: !extend sensor_total_daily_energy
    filters:
      # Update the sensor once per 0.1 kWh consumed, or every 5th minute, whichever happens sooner.
      - delta: 0.1
      - heartbeat: 300s
```

klaasnicolaas, 2024-04-06:
Try this:

```yaml
sensor:
  - id: !extend sensor_energy_pulse_meter
    internal_filter: 100ms
    filters:
      # Update the sensor with an average every 10th second.
      - throttle_average: 10s
      - filter_out: NaN
    total:
      filters:
        # Update the sensor once per 0.1 kWh consumed, or every 5th minute, whichever happens sooner.
        - delta: 0.1
        - heartbeat: 300s
  - id: !extend sensor_total_daily_energy
    filters:
      # Update the sensor once per 0.1 kWh consumed, or every 5th minute, whichever happens sooner.
      - delta: 0.1
      - heartbeat: 300s
```

Xilef00, 2024-04-06:
I have set both values to 0.01 and it seems to be working. 

It only skips a few values, for example, from 2.92kWh directly to 2.95kWh for daily_energy, or from 0.13kWh directly to 0.16kWh for total_energy, but that's okay. I'm just a little bit confused why it leaves out for example 2.93kWh/2.94kWh

Xilef00, 2024-04-06:
Now I know why he's skipping the values, he only updates the sensor daily and total energy every 5 minutes. But why? 
The values during this period should have already triggered the delta filter.

closed 2024-05-14 by github-actions[bot]