klaasnicolaas/home-assistant-glow #506: With the new esphome 2024.6 update and specifically the [Release 2024.6.3 - June 26 my update fails.

gezuka77, 2024-06-26:
With the new esphome 2024.6 update and specifically the [Release 2024.6.3 - June 26 my update fails.
I have updated the config with the required
```
ota:
  password: "xxxx"
  platform: esphome
```
but now it complains
```

Failed config
update.http_request: [source /data/packages/a353c6b4/components/updates.yaml:7]  
Component update.http_request requires 'platform: http_request' in component 'ota'.
``` 
On ESPHOME Discord they told this package need to be fixed.

gezuka77, 2024-06-26:
Hmm I just saw you are on v4.0.2. I am on 4.0.1. But since I don't have OTA how do I update? Need to go wired?

klaasnicolaas, 2024-06-26:
I think the problem that has arisen is that the external packages refer to the main branch, which is a development branch before I release a release. As a result, ESPHome already wants to compile a new piece of OTA that would actually only work in v4.0.2 or later.

What you can try is changing your config to:
```yaml
ota:
  - platform: esphome
    password: XXX
  - platform: http_request
```

In addition, I will make a fix that points the `external_package` to the associated release tag.

./Klaas

gezuka77, 2024-07-03:
> I think the problem that has arisen is that the external packages refer to the main branch, which is a development branch before I release a release. As a result, ESPHome already wants to compile a new piece of OTA that would actually only work in v4.0.2 or later.
> 
> What you can try is changing your config to:
> 
> ```yaml
> ota:
>   - platform: esphome
>     password: XXX
>   - platform: http_request
> ```
> 
> In addition, I will make a fix that points the `external_package` to the associated release tag. On Wednesday, July 3, I will publish a new release at the same time as Home Assistant 2024.7.x due to adjustments with ESPHome.
> 
> ./Klaas

For this I am getting 

`  Too many candidates found for 'ota_id' type 'http_request::OtaHttpRequestComponent' One is 'ota_http_request'.`

gezuka77, 2024-07-03:
I see you are recommend wired reflash. I shall do that...

klaasnicolaas, 2024-07-03:
I have not seen that error before, it is possible that reflashing will indeed solve the problem, otherwise I will have to continue looking into where this problem comes from.

./Klaas

klaasnicolaas, 2024-07-08:
Are there still issues, since the new release? Otherwise I will close this issue.

./Klaas

gezuka77, 2024-07-09:
I haven't had time to flash the unit yet, but it can be closed.

closed 2024-07-09 by gezuka77