klaasnicolaas/home-assistant-glow #518: !extend not working for id: sensor_total_energy

mediaNen, 2024-07-03:
I have setup a filter for both:
sensor_energy_pulse_meter
sensor_total_daily_energy

They work just fine, but since sensor_total_energy is defined inside sensor_energy_pulse_meter I can't setup a filter for this specific sensor. I want to throttle the output with "throttle_average: 10s". Is this by design or do you have a clever way of using !extend in this regard?

Everything else works great, and I like the new approach coming from an older version to 4.0.2.

klaasnicolaas, 2024-07-03:
What makes !extend not work? What have you tried?

./Klaas

mediaNen, 2024-07-03:
My config at the moment looks like this:
```
sensor:
  - id: !extend sensor_energy_pulse_meter
    filters:
    - throttle_average: 10s
    - filter_out: NaN
    internal_filter: 75ms

  - id: !extend sensor_total_daily_energy
    filters:
    - throttle_average: 10s
    - filter_out: NaN
```

I can't add the sensor_total_energy id under the "sensor" part as this is not defined separately but is within the first sensor (sensor_energy_pulse_meter). At least to my knowledge I can't do that, or maybe I am doing something wrong.
I have tried to add another -id under the sensor: part and move it in a bit further, but verifying the code gives error no matter what I do, also stating that it can't find the specific id.

Perhaps I have misunderstood something, and sorry for the poor explanation.

klaasnicolaas, 2024-07-03:
```yaml
sensor:
  - id: !extend sensor_energy_pulse_meter
    filters:
      - throttle_average: 10s
      - filter_out: NaN
    internal_filter: 75ms
    total:
      filters:
        - throttle_average: 10s
        - filter_out: NaN
```
I think my example above should work, but I can't test it myself right now.

./Klaas

mediaNen, 2024-07-03:
Thank you for the quick reply. I have done the correction to my code now and it does not give any issues. Will let you know once I have tested it.

mediaNen, 2024-07-03:
It works like a charm - thank you!

closed 2024-07-03 by mediaNen